MiCode/Xiaomi_Kernel_OpenSource #688: repo init: manifest not available for land kernel

MPeti1, 2017-07-20:
I wanted to prepare editing the kernel of Redmi3S (land), and i followed [this guide](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-To-Use#download-qualcomm-android-enablement-project) to do that.
I came across a problem at downloading QAEP.
As i have seen in the example, in `repo init` command i need to specify the kernel's release tag as the manifest, which i found with `git log` after `git checkout land-m-oss`.
so it seems i need to run this command:
`repo init -u git://codeaurora.org/platform/manifest.git -b release -m LA.UM.5.1-03810-8x37.0 --repo-url=git://codeaurora.org/tools/repo.git --repo-branch=caf-stable`
but after it downloads a lot of files it outputs this:
`Checking out files: 100% (16097/16097), done.`
`fatal: manifest 'LA.UM.5.1-03810-8x37.0' not available`
`fatal: manifest 'LA.UM.5.1-03810-8x37.0' not found`
But then i checked and it downloaded it. Then why it says it's not found? Is this a problem, or it's ok and i can proceed?
` * [new tag]           LA.UM.5.1-03810-8x37.0                                   -> LA.UM.5.1-03810-8x37.0`

[Here](https://gist.github.com/MPeti1/80aebcf078b1c321c7d58af8160b28c2) is the `git log` output of land kernel

luca020400, 2017-07-20:
Oh come on, you can't read again, compare the repo init command and you'll find out

MPeti1, 2017-07-20:
Now i already turned off my pc, but i think maybe the release branch could be a problem. I have already seen that before i asked here, but I'm not sure if this is the problem. Redmi 3S has msm8937 chipset, and at codeaurora's page you can see the nearest compatible chipset is msm8930, and if it can't be stated as 893x, this will explain the problem. But i searched for 8937 on this page, and not found any compatible branches for it. Anyway i will try again tomorrow (today, just later :D )

_Sent from my Xiaomi Redmi 3S using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

luca020400, 2017-07-20:
The closest to msm8937 is msm8953 and msm8996

MPeti1, 2017-07-21:
But there is also 8930. Or where do you see it? I'm searching again in the wrong place? :D 
And then what branch should i use if no one supports exactly 8937?
[image]

MPeti1, 2017-07-21:
I started to read `repo init --help`, and really the problem is i forgot to write .xml to the end of the manifest? :D

MPeti1, 2017-07-21:
Now it's working, really i can't read, thanks :D

MPeti1, 2017-07-23:
I came to another problem using `repo`
First i tried to download everything as the guide wanted, but my storage is filled up, and i needed another method to download only what i need. I searched for what can i do, and came to that:
First i added that to the end of repo init command: `--depth=1 --groups=all,-notdefault,-device,-darwin,-x86,-mips,-exynos5`
Then when it finished, i ran repo sync this way: `repo sync -c -n -j 2`
repo sync started, but it always gets stuck at `Fetching project platform/packages/apps/QuickSearchBox`
Last time i left my pc to do what it need, because i've seen it has network traffic while repo sync is stuck where i mentioned (traffic gone if i exited repo sync), but after about 1.5 hour i came back and the terminal wrote some runtime exceptions mentioning some python libs.
What could be the problem?

MPeti1, 2017-07-27:
fixed with setting depth to 0, and removing --groups

closed 2017-07-27 by MPeti1