MiCode/Xiaomi_Kernel_OpenSource #692: Respect the GNU GPLv2!!

amarpreetosahan, 2017-08-01:
Linux is distributed under the GNU GPLv2 license, you can not distribute software without source! Xaiomi have violated the GNU GPLv2 with the distribution of Miui 8 (5.1.1) and miui 8.5 (6.0)!
I will contact the copyright owner (Linus Torvalds) to report the repeated violation of the license by Xaiomi

Thanks for the attention

luk1337, 2017-08-01:
>I will contact the copyright owner (Linus Torvalds)
lmao

TheStrix, 2017-08-01:
Lmao

marines, 2017-08-01:
Nothing to laugh about. It's not as stupid as Xiaomi not complying with GPL.

amarpreetosahan, 2017-08-01:
@TheStrix as a developer i respect you a lot, but is this really thing of laughs?

luk1337, 2017-08-01:
@amarpreetosahan yes indeed.

amarpreetosahan, 2017-08-01:
@luk1337 don't you think that xaiomi have to follows the license from they arw selling his product?

luk1337, 2017-08-01:
@amarpreetosahan we weren't laughing at that, we were laughing at you.

luk1337, 2017-08-01:
Also it looks like you are unable to spell Xiaomi right. Just FYI it's X I A O M I.

amarpreetosahan, 2017-08-01:
Yeah yeah yeah @luk1337 you people can laughs on me as i didn't write xiaomi 's spell correctly or i said that i will tag the post to owner, bur can't you see the issue.

akhilnarang, 2017-08-01:
🤣

nealgosalia, 2017-08-01:
In maymay land, we say it as,
"one does not simply accuse Xiaomi of not releasing sources"

marines, 2017-08-02:
@luk1337 you're dumb.

ohjames, 2017-08-19:
Xiaomi are a bunch of tossers, they don't care about the GPL. Don't buy their products. Shame as my next phone would definitely be Xiaomi if they weren't such greasy bastards.

zieru, 2017-09-08:
1 month already, did you already contacted linus? :D

out386, 2017-09-24:
> I will contact the copyright owner (Linus Torvalds)

Just one single owner? :rofl:

935713827, 2017-09-25:
在China  何为GNU GPLv2

kiodo1981, 2017-10-11:
I hope and soon release source for Redmi 4A (rolex)
https://www.change.org/p/xiaomi-inc-publish-the-kernel-source-for-the-xiaomi-redmi-4a-4-4pro

lulujyc, 2017-10-12:
There's no really restricts for Open-Source terms in China now :(

ohjames, 2017-10-12:
Do you open a change petition when a drug dealer rips you off in the street when you're buying weed? And yet you open one for the criminals at Xiaomi? LOL.