MiCode/Xiaomi_Kernel_OpenSource #698: Release the Kernel Source for Xiaomi Redmi 4X (Santoni)

jawaadsaeed, 2017-08-13:
Hi, please hear us, we need Kernel Source for Santoni and i want to do a development for this device. Redmi 4X is a great phone now and so many people using and loving it! Redmi Note 4X has 4 months for waiting the Kernel Source and you released the Kernel Source for it. And for Redmi 3 too. Then, why you must delayed Kernel Source for Xiaomi Redmi 4X? Its been 5 months we've been waiting for Kernel Source release! Please hear us Xiaomi! You are the best!

Thank you!

luca020400, 2017-08-13:
Tip: Xiaomi is not the best, just because you're here.

luk1337, 2017-08-13:
5 months? As a Xiaomi fan you can way up to 18 months ^.^

luca020400, 2017-08-13:
Best i can do is 6 months.

notly, 2017-08-14:
Make it be never!!!

jawaadsaeed, 2017-08-14:
I want to make Custom Kernel!!!!!

luca020400, 2017-08-14:
Add one more ! and the genius of the kernel will appear.

jawaadsaeed, 2017-08-14:
SHAME XIAOMI IF THEY DIDN'T RELEASE REDMI 4X KERNEL SOURCE!!!

luk1337, 2017-08-14:
!!!!!!!!!!!!!!!!!!!!!!!!!!!

luca020400, 2017-08-14:
Now you have 3 wishes, do wise choices.

ohjames, 2017-08-19:
You shouldn't have bought a phone made by the IP thieves at Xiaomi.

jawaadsaeed, 2017-08-19:
I don't need you here, so GTFO ,,

luk1337, 2017-08-19:
lmao~

jawaadsaeed, 2017-08-19:
I've made this issues for Xiaomi, not for you James ,,

luk1337, 2017-08-19:
@jawaadsaeed why do you always put space and two commas at the end of your sentences? What's the point of it?

ohjames, 2017-08-19:
You've kinda missed my point: Xiaomi have been violating the GPL for years, they are criminals who don't care about international copyright law. Your github issue is insignificant to them, pissing in the wind. The only way to get corporate monsters to change is to hit them in their wallets i.e. research your products and don't buy from criminals. Xiaomi are as likely to listen to you after you've been conned by them as a fake dealer is gonna give someone their money back after conning them. I know, getting hit by a criminal is not a nice feeling, but you gotta learn from it and get a bit more cynical.

ohjames, 2017-08-19:
There's a reason you had to buy your phone from some shitty internet third party reseller or the criminal's own "official" website.

The same way there's a reason why the dealer is yelling out "hot bud" next to a crowded train station and selling you "ganja" in a black bag (that's really parsley) for 20 dollars.

jawaadsaeed, 2017-08-19:
Thanks for that @ohjames
I understand now ,,

ohjames, 2017-08-19:
What is it with the " ,, " do you just enjoy looking like an 5 year old who just learned both internet and punctuation or something? Is that really gonna help Xiaomi take you seriously (or anyone else for that matter). Lol.

jawaadsaeed, 2017-08-19:
Don't take it too seriously lmao

ohjames, 2017-08-19:
We put up with your pointless insults and whining, we deserve to know.

jawaadsaeed, 2017-08-19:
Lmao xD

SwapnilSoni1999, 2017-10-19:
we are here to hope that xiaomi 'may' release kernel source ...

and xiaomi be like: lmao dude ... what are you talking about...

Lanados, 2017-10-31:
I understand it will not open the source for redmi 4x

jawaadsaeed, 2017-10-31:
Yeaa

Lanados, 2017-10-31:
Bad

Lanados, 2017-10-31:
whence such information

Lanados, 2017-11-01:
Please, xiaomi, open kernel sources😭😭😭😭😭😭😭😭😭

jawaadsaeed, 2017-11-02:
Don't take too much hope on Xiaomi

shatanid, 2017-11-22:
please Release source for redmi 4x

shatanid, 2017-11-22:
huh -_-

julian-alarcon, 2017-11-22:
The new MIUI v9 is near to be released, and still no Source Code of Kernel. If they begin to sell Redmi 4X phones in Spain they should respect the GPLv2 license and release the code.

notly, 2017-11-24:
XiaoMi: GPL? No Way!!!

jawaadsaeed, 2017-11-25:
Xiaomi is an esshole lol

Riora, 2017-12-05:
HEY XIAOMI! it's a GPLv2 violation!!!

luk1337, 2017-12-05:
cool

Lanados, 2017-12-14:
I will never again buy a smartphone xiaomi with such an irresponsible developer, you are a fagot, burn in hell small Chinese with small members

luk1337, 2017-12-14:
lmao

jawaadsaeed, 2017-12-14:
Hehehe

ryan-andri, 2017-12-14:
lul

jawaadsaeed, 2017-12-14:
@ryan-andri im testing your kernel with unofficial source bro

ryan-andri, 2017-12-14:
@jawaadsaeed sure, good luck mate :+1:

julian-alarcon, 2018-01-18:
Related bugs:
- https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/787
- https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/767

Somebody should merge them. I guest main is 698 (older and more comments).

julian-alarcon, 2018-01-18:
By the way, the are didn't release the kernel source for Xiami Mi A1 an Android One phone!
http://www.androidpolice.com/2018/01/16/xiaomi-still-hasnt-released-kernel-sources-mi-a1/
https://arstechnica.com/gadgets/2018/01/xiaomi-is-dragging-its-feet-on-the-gpl-again-this-time-with-the-mi-a1/

luca020400, 2018-01-18:
Say hello to the new kernel issues sheriff: sir @julian-alarcon.

jawaadsaeed, 2018-02-26:
Issues closed! Source already released today!

closed 2018-02-26 by jawaadsaeed

julian-alarcon, 2018-02-26:
Source code finally released:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/09baf1a3dd0e13a81e484035f0be71dcb2ecc723