MiCode/Xiaomi_Kernel_OpenSource #701: Version script assignment of LIBC to symbol __propClientDispatch failed (Redmi Note 4X mido)

veiz, 2017-08-23:
Hi, when I try to build Redmi Note 4X mido-m-oss kernel with QAEP `aarch64-linux-android-4.9` toolchain, I encountered a LD problem as follows, someone can help? 

```
prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/aarch64-linux-android/bin/ld.gold: error: version script assignment of LIBC to symbol __propClientDispatch failed: symbol not defined
prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/aarch64-linux-android/bin/ld.gold: error: version script assignment of LIBC to symbol __propClientDispatchWrite failed: symbol not defined
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
build/core/ninja.mk:148: recipe for target 'ninja_wrapper' failed
make: *** [ninja_wrapper] Error 1
```

ogsts, 2017-12-05:
Check if BoardConfig.mk has
BOARD_USES_LIBC_WRAPPER := true

ryan-andri, 2017-12-05:
change ld gold to bfd

triplekiller, 2017-12-09:
@ogsts 我也遇到这个问题，怎么解决的？？？

Starting build with ninja
ninja: Entering directory `.'
[  0% 8/914] target thumb C: libcutils <= system/core/libcutils/socket_network_client_unix.c
[  0% 9/914] target thumb C: libcutils <= system/core/libcutils/str_parms.c
[  1% 10/914] target thumb C: libcutils <= system/core/libcutils/klog.c
[  1% 11/914] target SharedLib: libc (out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/LINKED/libc.so)
FAILED: /bin/bash -c "prebuilts/clang/host/linux-x86/clang-2690385/bin/clang++ -nostdlib -Wl,-soname,libc.so -Wl,--gc-sections -shared  -Lout/target/product/msm8909/obj/lib    out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbrand.o out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtend_so.o   out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/bionic/malloc_common.o out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/bionic/libc_init_dynamic.o out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/bionic/NetdClient.o       out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/arch-common/bionic/crtbegin_so.o out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/arch-arm/bionic/exidx_dynamic.o out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/arch-arm/bionic/atexit_legacy.o        -Wl,--whole-archive  out/target/product/msm8909/obj/STATIC_LIBRARIES/libc_common_intermediates/libc_common.a out/target/product/msm8909/obj/STATIC_LIBRARIES/libjemalloc_intermediates/libjemalloc.a -Wl,--no-whole-archive   out/target/product/msm8909/obj/STATIC_LIBRARIES/libcompiler_rt-extras_intermediates/libcompiler_rt-extras.a   prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/bin/../lib/gcc/arm-linux-androideabi/4.9/../../../../arm-linux-androideabi/lib/libatomic.a prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/bin/../lib/gcc/arm-linux-androideabi/4.9/libgcc.a -ldl  -o out/target/product/msm8909/obj/SHARED_LIBRARIES/libc_intermediates/LINKED/libc.so   -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -Wl,--build-id=md5 -Wl,--warn-shared-textrel -Wl,--fatal-warnings -Wl,--icf=safe -Wl,--hash-style=gnu -Wl,--no-undefined-version -Wl,--no-fix-cortex-a8    -target arm-linux-androideabi -Bprebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/arm-linux-androideabi/bin   -Wl,--hash-style=both -Wl,--version-script,bionic/libc/libc.arm.map -Wl,--no-undefined"
prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/arm-linux-androideabi/bin/ld: error: version script assignment of LIBC to symbol __propClientDispatch failed: symbol not defined
prebuilts/gcc/linux-x86/arm/arm-linux-androideabi-4.9/arm-linux-androideabi/bin/ld: error: version script assignment of LIBC to symbol __propClientDispatchWrite failed: symbol not defined
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
[  1% 11/914] target thumb C: libcutils <= system/core/libcutils/debugger.c
[  1% 11/914] target thumb C: libcutils <= system/core/libcutils/android_reboot.c
[  1% 11/914] target thumb C: libcutils <= system/core/libcutils/partition_utils.c
[  1% 11/914] target thumb C: libcutils <= system/core/libcutils/ashmem-dev.c
[  1% 11/914] target thumb C: libcutils <= system/core/libcutils/properties.c
[  1% 11/914] target thumb C: libcutils <= system/core/libcutils/trace-dev.c
[  1% 11/914] build out/target/product/msm8909/obj/KERNEL_OBJ/usr
/bin/bash: @: command not found