MiCode/Xiaomi_Kernel_OpenSource #765: Release kernel sources for Xiaomi Mi A1

ravinder0003, 2017-09-14:
Please release kernel sources for Mi A1 asap.
Thanks

Aniketh-Raj, 2017-09-19:
Hello Xiaomi, Please release the Kernel Sources for Mi A1 TISSOT.. The phone was released some time back and still Kernel Sources are not up even when the device is being called as Android ONE...!! Please release the sources asap

subhangjoshi16, 2017-09-22:
Give M1 a1 sources even if it is Android one device then also no source not released till now

kostoulhs, 2017-10-01:
Release Mi A1 kernel sources please

ves626, 2017-10-02:
Xiaomi, please _release Mi A1 kernel_ ASAP, devs are still waiting...

ashishranjan738, 2017-10-05:
Xiaomi please the release the kernel sources ASAP

kunalshah912, 2017-10-06:
Come ON Xiaomi ! You are not as bad as OnePlus is "with handling Customers!" Release the Kernel source for Mi A1....

rickling1993, 2017-10-07:
Please release Mi A1 kernel sources. It is a shame that an Android One device by Google partnership didn't follow GPL requirement.

DamandeepS, 2017-10-07:
Please provide the source code.

kostoulhs, 2017-10-10:
I contacted Xiaomi Global Support and they replied that they are gonna forward my request to the appropriate department of Xiaomi. Just waiting, for a couple of time for now.

luk1337, 2017-10-11:
no~

marines, 2017-10-11:
@luk1337

luca020400, 2017-10-11:
@marines

TechExhibeo, 2017-10-11:
hakuna your tatas

luca020400, 2017-10-11:
Swahili on github, that's new.

sachoosaini, 2017-10-18:
Please release kernel sources for XIAOMI, I bought this thinking this is google android one device and source code will be available as soon as this phone is launched after few days.  Even one of elite developers of LINEAGE is here I can see that which makes me happy.

TechExhibeo, 2017-10-18:
^ Sure thing, till evening it should be resolved :D
And thanks for reminding :)

percy-g2, 2017-10-19:
Please provide the source code.

Aymahalawa, 2017-10-19:
Pls release the Mi A1 kernel source , This is Android one phone !!

jaikrishna835, 2017-10-30:
https://www.change.org/p/people-xiomi-mi-a1-kernel-source?recruiter=677404613&utm_source=share_petition&utm_medium=copylink&utm_campaign=share_petition&utm_term=share_for_starters_page

jaikrishna835, 2017-10-30:
Sign the above petition friends

bennyhillthebest, 2017-11-05:
Please release the source code Xiaomi. The Mi A1 is the nexus successor

luk1337, 2017-11-05:
successor? I think you have no idea what you're talking about.

hugwalk, 2017-11-08:
大家都等很久了(
Source 就快给吧…
Devs 都没办法开发

linpapa, 2017-11-09:
hi, please release the sourse code, it is no harm

HenrikNyberg, 2017-11-10:
You have released Kernel source to other phones even in the redmi series, this is a AndroidOne, what is taking so long, please respect GPL2! This is my first Xiaomi please don't make it my last..

luca020400, 2017-11-11:
FYI: theoretically, google should handle android one kernel releases.

abler98, 2017-11-13:
I already really want to start developing under this device, how long to wait for the kernel source?

sachoosaini, 2017-11-14:
XIAOMI common what is taking so long. I am now regretting purchasing this crap device seriously. Your own ROM is bugged with so many issues. Why isn't google listening to these comments? HTC and MOTOROLA are way better at handling things when it comes to developers. You proved again and again that why I forced people in the past not to buy your devices. There is no respect for GPL. You proved me right. I am regretting that I Buy your device.

roberthchan, 2017-11-18:
Release Please.

asrupesh, 2017-11-18:
Hello Xiaomi, Please release the Kernel Sources for Mi A1.. The phone was released more than 2 months and still Kernel Sources are not up even when the device is being called as Android ONE...!! Please release the sources ASAP.......

adeveloper86, 2017-11-18:
Hello Xiaomi, Please release the Kernel Sources for Mi A1.. The phone was released more than 2 months and still Kernel Sources are not up even when the device is being called as Android ONE...!! Please release the sources ASAP.......

sh4hidkh4n, 2017-11-18:
What's taking too much time? One plus released kernel source and device for 5T within days. Much disappointed

DeicPro, 2017-11-18:
You are requited by GPLv2 license to release source code, just do it!

scafroglia93, 2017-11-18:
CIAO MAMMA

luca020400, 2017-11-18:
SONO IN TV!1!!1!1!11

percy-g2, 2017-11-18:
[image]

zeroxxx2, 2017-11-19:
Bunch of crying babies. Be patient or GTFO.

sachoosaini, 2017-11-19:
At least now wake up 
https://www.xda-developers.com/xiaomi-not-released-kernel-sources-mi-a1/

Kizoky, 2017-11-19:
@zeroxxx2 Sure mate. I'm still waiting for Redmi 4 prada source code to be released, and this is still not done after 1 year.

nilmarsh555, 2017-11-20:
Shame on you Xiaomi. Kernel source code for one plus 5x released on first day of launch.

mohanen, 2017-11-21:
Still no Xiaomi_Kernel_OpenSource Maintainers have replied to this thread . Just Great

acnalogia, 2017-11-25:
Xiaomi, please release Mi A1 kernel ASAP

scafroglia93, 2017-11-25:
I don't understand, you know that Xiaomi behaves this way and you continue to buy it

sachoosaini, 2017-11-25:
@scafroglia93 see even now they are doing this. I mean this company never respected GPL in the past too what can be expected now?

sh4hidkh4n, 2017-11-29:
No fucking kernel source yet.. its been 3 months to this issue creation. How much time more?

luk1337, 2017-11-29:
As much as they like :^)

mohanen, 2017-11-29:
https://www.gnu.org/licenses/gpl-violation.html
license-violation@gnu.org

zeroxxx2, 2017-11-29:
Yeah right, when it comes to a violation if it's really punished almost nobody would be able to produce Android devices anymore. Nobody gets hurt even if they don't release it.

Suck it up, wait for them or move on.

mohanen, 2017-11-29:
@zeroxxx2  dude nobody asked for your advice, move on

zeroxxx2, 2017-11-29:
Dude I can be anywhere, do you pay for my internet connection? Suck it up and move on.

ashu00, 2017-12-04:
https://github.com/cuteprince/android_kernel_xiaomi_msm8953

ishantvivek, 2017-12-05:
I think, he ported kernel ^ based on kumajaya guide, like he did for Redmi 2.

ashu00, 2017-12-05:
^ Yes, he is porting from redmi note 4.

KonradIT, 2017-12-06:
You're violating the GPL, if you sell the Mi A1 in the US you can get sued.

dong-dev, 2017-12-14:
15 days before the end of year

luk1337, 2017-12-14:
cool

ishantvivek, 2017-12-14:
yup very cool weather here^ , how do you come to know? 🤔🤔😂😂

sh4hidkh4n, 2017-12-14:
So cool that my hands are freezing..

deyjishnu, 2017-12-25:
Please release the kernel source

scafroglia93, 2017-12-25:
Merry Christmas

[19 more replies]