MiCode/Xiaomi_Kernel_OpenSource #767: Kernel Sources for Redmi 4x Santoni

ramnikhilraju19, 2017-09-15:
Dear Xiaomi, you are bound to follow and respect the GPL 2 and release the kernel Sources for all of the devices you launch
So do release the sources for Redmi 4x Santoni, which was released almost a year ago.

julian-alarcon, 2018-01-18:
Related bugs:
- https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/698
- https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/787

Somebody should merge them. I guest main is 698 (older and more comments).

julian-alarcon, 2018-01-18:
By the way, the are didn't release the kernel source for Xiami Mi A1 an Android One phone!
http://www.androidpolice.com/2018/01/16/xiaomi-still-hasnt-released-kernel-sources-mi-a1/
https://arstechnica.com/gadgets/2018/01/xiaomi-is-dragging-its-feet-on-the-gpl-again-this-time-with-the-mi-a1/

julian-alarcon, 2018-02-26:
Please (finally) close this bug, source code released:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/09baf1a3dd0e13a81e484035f0be71dcb2ecc723