MiCode/Xiaomi_Kernel_OpenSource #776: [MI6] Camera doesn't work with OSS kernel

TheStrix, 2017-09-30:
Crash Log:
```
[    1.263183] msm_actuator_platform_probe:1957 msm_actuator_platform_probe: 1957: Actuator no GPIO control
[    1.263227] msm_actuator_platform_probe:1957 msm_actuator_platform_probe: 1957: Actuator no GPIO control
[    1.263661] msm_eeprom_platform_probe qcom,eeprom-name sagit_imx386_semco, rc 0
[    1.263800] qcom,slave-addr = 0xA0
[    1.267209] msm_cci_init:1380: hw_version = 0x10060000
[    1.533852] msm_eeprom_platform_probe: sensor_eeprom_name[0] = sagit_imx386_semco, probe success!
[    1.534306] msm_eeprom_platform_probe qcom,eeprom-name sagit_s5k3m3_semco, rc 0
[    1.534502] qcom,slave-addr = 0xA0
[    1.538051] msm_cci_init:1380: hw_version = 0x10060000
[    1.723881] msm_eeprom_platform_probe: sensor_eeprom_name[1] = sagit_s5k3m3_semco, probe success!
[    1.724272] msm_eeprom_platform_probe qcom,eeprom-name sagit_imx268_primax, rc 0
[    1.724731] qcom,slave-addr = 0x34
[    1.741865] synaptics_rmi4_fwu_init: fwu->in_ub_mode = 0
[    1.743701] msm_cci_init:1380: hw_version = 0x10060000
[    1.743744] qcom,slave-addr = 0x34
[    1.760186] synaptics_dsi_force 5-0020: fwu_read_f34_guest_serialization_partition: Lockdown info: 0x31 0x31 0x32 0x2 0x43 0x31 0x31 0x0
[    1.761320] synaptics_dsi_force 5-0020: fwu_get_firmware_name: Choose firmware synaptics_jdi_3330_c1.fw
[    1.762719] qcom,slave-addr = 0x34
[    1.774593] synaptics_dsi_force 5-0020: fwu_start_reflash: Start of reflash process
[    1.775465] synaptics_dsi_force 5-0020: fwu_go_nogo: Device firmware ID = 2598829
[    1.775474] synaptics_dsi_force 5-0020: fwu_go_nogo: Image firmware ID = 2598829
[    1.777020] synaptics_dsi_force 5-0020: fwu_go_nogo: No need to do reflash
[    1.777028] fwu_start_reflash: End of reflash process
[    1.782690] qcom,slave-addr = 0x34
[    1.802686] qcom,slave-addr = 0x34
[    1.829364] msm_eeprom_platform_probe: sensor_eeprom_name[2] = sagit_imx268_primax, probe success!
[    1.829821] msm_eeprom_platform_probe qcom,eeprom-name sagit_imx268_liteon, rc 0
[    1.830296] qcom,slave-addr = 0x34
[    1.849155] msm_cci_init:1380: hw_version = 0x10060000
[    1.849192] qcom,slave-addr = 0x34
[    1.862697] qcom,slave-addr = 0x34
[    1.882686] qcom,slave-addr = 0x34
[    1.902687] qcom,slave-addr = 0x34
[    1.929181] qcom,eeprom: probe of ca0c000.qcom,cci:qcom,eeprom@3 failed with error -12
[    1.930057] msm_ois_platform_probe:994 msm_ois_platform_probe: 994: OIS no GPIO control
[    1.933093] failes: no ois node, maybe didn't support ois
[    1.933784] failes: no ois node, maybe didn't support ois
```

I see SENSOR_SUPPORT_OIS_FLAG failure. 

@liuguo09

luca020400, 2017-09-30:
Don't expect xiaomi camera to work out of the box, you have to fix it \o/

TheStrix, 2017-11-02:
Was the issue with blobs compatibility, should be fixed with newer kernel release.

closed 2017-11-02 by TheStrix