MiCode/Xiaomi_Kernel_OpenSource #777: Please release kernel sources for MI MAX 2 - oxygen

SubZeroNexii, 2017-10-01:
I have been waiting for a while now. The Mi 6 sources have already been released and this phone has been released in the same period with it. I kindly ask you to release the kernel sources as soon as possible. Thank you.

sathwikrao575, 2017-10-03:
Yes

luk1337, 2017-10-03:
No

luca020400, 2017-10-03:
Maybe

luca020400, 2017-10-03:
Just kidding, no

devesh38, 2017-10-06:
please release Kernel sources of Mi Max 2

neerajvohra, 2017-10-07:
@luca020400 What's the usual time for Xiaomi to release kernel sources? 6 months right? \0/

eliaspizarro, 2017-12-20:
3 years

reversegear, 2017-12-22:
Xiaomi please release the source code for Mi Max 2

luk1337, 2017-12-22:
why?

Darghoon, 2017-12-22:
Release kernel sources, please...

luca020400, 2017-12-22:
soon™

andrea2107, 2017-12-22:
Gib sources plox

luk1337, 2017-12-22:
No.

andrea2107, 2017-12-22:
U're rude :(

luca020400, 2017-12-22:
*You're

andrea2107, 2017-12-22:
No, u're

Shish

luca020400, 2017-12-22:
Shut up Andrea.

andrea2107, 2017-12-22:
No u

DrWssn, 2017-12-23:
Plz realase the kernal xiaomi , we need this

mafcarvalho, 2018-01-02:
People at Xiaomi, please release the kernel source code for Mi Max 2 immediately.
You need to do it in order to comply with GPLv2. 
Please note that doing it will increase sales of the device. A good smartphone with open source development is what we need. 
More than 6 months passed since the release of this great device. Why don't you help us and yourselves by doing the right thing? 
Thanks in advance!

slyscorpion, 2018-01-08:
Looks like they released the kernel sources today 

https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/oxygen-n-oss?files=1

mafcarvalho, 2018-01-08:
@slyscorpion thanks a lot for the information! Kudos

andrea2107, 2018-01-08:
I guess the spam worked.