MiCode/Xiaomi_Kernel_OpenSource #778: Release kernel source for Mi Max 2

sathwikrao575, 2017-10-03:
Unofficial Lineage OS Development stopped because Xiaomi didn't release kernel
XDA closed thread of Unofficial Lineage OS stating reason as "Thread closed due to GPL." . This ROM was based on Redmi Note 4 Kernel.
This thread was closed because Xiaomi didn't release Kernel sources of Mi Max 2.
This is sad news for all those using this ROM as it won't be having further development.
Please release Kernel source of Mi Max 2 it's been about 5 months Since this phone was launched.
There is no explanation why sources are not released.
If you release Kernel sources, development of this phone will be more ,increasing sales.
So please release Kernel sources of Mi Max 2

devesh38, 2017-10-06:
please release Kernel sources of Mi Max 2

casterland, 2017-11-05:
Good hardware, let the developers take full advantage of it!

Pidus, 2017-12-03:
Release the source. It's using the same SD 625 that Redmi Note 4 uses. And still we have no kernel source. Why? Why are you all silent?

fahad86, 2017-12-05:
Bought the phone only by seeing the open-source support available for other Xiaomi phones like Redmi Note 4. Very disappointed. Will have to think twice before buying another Xiaomi device.

KapilTapsi, 2017-12-15:
please release it asap. we really need it

TiMbl4, 2017-12-25:
It's been almost seven(!) months, since the release date. By the looks of it, the kernel source will be released with the Max 3... -_-'

thananop1987, 2018-05-04:
good good