MiCode/Xiaomi_Kernel_OpenSource #787: Release Kernel Source code for Xiaomi Redmi 4X

jawaadsaeed, 2017-10-17:
Please Xiaomi release Kernel Source for Redmi 4X

bugerxXx, 2017-10-22:
Yes Pleasee...

peterius, 2017-10-26:
Still in progress but it's getting to init:
https://github.com/peterius/Xiaomi_Kernel_OpenSource_fbcon/tree/mido_fbcon

This is for the Note 4X though, but maybe it works as well?  There's all kinds of errors, even in the kernel provided with the phone.

jawaadsaeed, 2017-11-06:
@hehao

ryan-andri, 2017-11-24:
so xiaomi not gentle as big company !

bitrvmpd, 2017-12-08:
Hey, this one is based on CAF sources https://github.com/bitrvmpd/msm-3.18 =)

jawaadsaeed, 2017-12-08:
@bitrvmpd Yes bro we actually use your source for now on ❤ you are amazing

peterius, 2017-12-08:
What's working with the bitrvmpd CAF kernel?  Am I duplicating his efforts?  Will it work on a Redmi Note 4X, radio, etc.?

bitrvmpd, 2017-12-08:
@peterius everything is working (maybe for some people IR blaster won't work, but I need logs to fox that). And it's only for Redmi 4X (santoni)

sahroni28, 2017-12-20:
Just wait long

arizaini, 2017-12-27:
Goto https://forum.xda-developers.com/xiaomi-redmi-4x and u will find what u want

jawaadsaeed, 2017-12-27:
@arizaini We need official source to fix bugs on Oreo

arizaini, 2017-12-27:
What kind of bug?

semkir216, 2018-01-16:
NitrogenOS Oreo Stable released go to MIUI Forum visit my thread

luk1337, 2018-01-16:
fuck off

julian-alarcon, 2018-01-18:
Related bugs:
- https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/698
- https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/767

Somebody should merge them. I guest main is 698 (older and more comments).

julian-alarcon, 2018-01-18:
By the way, the are didn't release the kernel source for Xiami Mi A1 an Android One phone!
http://www.androidpolice.com/2018/01/16/xiaomi-still-hasnt-released-kernel-sources-mi-a1/
https://arstechnica.com/gadgets/2018/01/xiaomi-is-dragging-its-feet-on-the-gpl-again-this-time-with-the-mi-a1/

jawaadsaeed, 2018-02-26:
Issues closed! Source already released today!

closed 2018-02-26 by jawaadsaeed

julian-alarcon, 2018-02-26:
Source code finally released: 09baf1a