MiCode/Xiaomi_Kernel_OpenSource #789: Can't build kernel source of MI MIX. build failed.

liuchibing, 2017-10-22:
I got the following error message when compiling the kernel source in branch "scorpio-m-oss":

> $ make
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CALL    scripts/checksyscalls.sh
  CHK     include/generated/compile.h
  CC      arch/arm64/kernel/bootinfo.o
In file included from arch/arm64/kernel/bootinfo.c:19:0:
include/linux/qpnp/power-on.h:76:12: warning: ‘qpnp_pon_system_pwr_off’ defined but not used [-Wunused-function]
error, forbidden warning: power-on.h:76
scripts/Makefile.build:257: recipe for target 'arch/arm64/kernel/bootinfo.o' failedmake[1]: *** [arch/arm64/kernel/bootinfo.o] Error 1
Makefile:945: recipe for target 'arch/arm64/kernel' failed
make: *** [arch/arm64/kernel] Error 

Is it a bug? Is there any solution? Or can I just ignore it?

liuchibing, 2017-10-22:
After that I found another 4 errors...T^T Is there any method to successfully compile a kernel for Mi Mix?

ItzLevvie, 2017-10-22:
The kernel source code you're building is for Xiaomi Mi Note 2 (scorpio)