MiCode/Xiaomi_Kernel_OpenSource #792: Mi Note 3 kernel source

travismills82, 2017-10-30:
Where is it?

travismills82, 2017-10-30:
Is helping you to transfer manual service, please wait ...
Talk to Mike now
Hello, welcome to use millet Hong Kong online service. I'm Mike, is there anything that can help you? Such as busy lines may lead to slow response, thank you for your understanding! Thank you.
Hello, how can I help you?
#
I would like to request Xiaomi Mi Note 3 Kernel Source code
Under GPL License this should be available correct?
#
Hello, may I ask you to buy millet note3 on the mainland?
#
Hong Kong
there is not any source available where Xiaomi kernel source is published here  https://github.com/MiCode/Xiaomi_Kernel_OpenSource
#
Hello, the current Hong Kong area and no millet note3 in the sale, may I ask you through what channels to buy?
#
2017-10-30 16:02:28
Many stores in Kowloon sale this mobile
Kernel Source under GPL laws may I ask when this will be released
https://www.price.com.hk/product.php?p=276119
many sell in Hong Kong this is not an issue
Issue is NO KERNEL SOURCE when Xiaomi is required to release
Kernel source is not located on github here:  https://github.com/MiCode/Xiaomi_Kernel_OpenSource
#
We only accept products from Hong Kong's official channels (Hong Kong MI home, Hong Kong millet official website and CSL). Since there is no such product in Hong Kong, we suggest you can contact the seller. At present millet note3 is not available in Hong Kong, we do not have millet note 3 related information.
And can not provide you with the relevant services and technical support, thank you for your understanding.
2017-10-30 16:07:28
#
has nothing to do with device purchase
I will paste this conversation to Qualcomm and have them get back to Xiaomi for violating GPL law
thank you for your time and your misunderstanding of GPL LAW

### Now they are supposed to release sources correct and not just ignore this fact? ###

scafroglia93, 2017-10-31:
Buy another phone

jmcantos, 2017-12-02:
Hello, any news about when will Mi Note 3 source code be released?

travismills82, 2017-12-08:
There are better options with snap 660 cpu with kernel source released now so there is no need to purchase this mobile anymore.

jmcantos, 2017-12-08:
This phone has all that is needed to be the next Nexus 5 in terms of community support: nice specs per price and a famous brand. It only misses the kernel sources. It's sad.

Titokhan, 2017-12-22:
It's available now: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/jason-n-oss