MiCode/Xiaomi_Kernel_OpenSource #795: Redmi 4A (Rolex) Kernel Source

Sourav-21, 2017-11-02:
# Are you **fucking** kidding us?
## This mobile was released **1 year** ago and no marshmallow update no kernel source. 
What are you doing? What is the problem to upload those codes? 
### Is there any hidden module for spying on Indian users? Are you stealing our personal information and send them to China? Are you cheating in the benchmark?

NisseCool, 2017-11-02:
Spyware...

jawaadsaeed, 2017-11-03:
Its no use. Hehao is an idiot

darjiyo, 2017-11-08:
https://www.change.org/p/xiaomi-inc-publish-the-kernel-source-for-the-xiaomi-redmi-4a-4-4pro?utm_medium=email&utm_source=notification&utm_campaign=petition_signer_receipt&share_context=signature_receipt&recruiter=743375047

Nibi-git, 2018-01-06:
I join the question!

ryan-andri, 2018-01-07:
what -___-!

LuanHalaiko, 2018-01-09:
Tafuq? (-__-)

R o l e x (I can't even say how obscene this word is in my country)
And no, probably they won't release-it

turjyasaha, 2018-01-10:
+2

muralivijay, 2018-01-31:
please release xiaomi

maxismangh, 2018-03-07:
Still no kernel sources! This is ridiculous.

dineshdb, 2018-03-28:
#825 . Finally after so long. Xiaomi sucks!