MiCode/Xiaomi_Kernel_OpenSource #798: Release kernel source for Mi Max 2

123oleg1977, 2017-11-13:
You should at least have the approximate date of the kernel source code output.
With respect by your products.

123oleg1977, 2017-11-18:
Show respect to the people and tell them if the kernel source code is provided, everyone hopes, and the subject is not answered, from you. With respect.

123oleg1977, 2017-12-11:
I sold mi max2, due to the omission of the source code of the kernel, because I bought the device with the calculation of the LOS installation, and since you have xiaomi such attitude to customers in the future, I will not return to your products, very disappointed, good luck to you ..

devishell, 2018-01-12:
is this?
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/oxygen-n-oss?files=1