MiCode/Xiaomi_Kernel_OpenSource #802: source codes for Redmi Note 4 MTK for the love of God

Ayrtown-Karlos, 2017-12-07:
If you do not want to update it, let the Roms developers update it,
Vernee Apollo Lite uses Helio X20 and has source code available,
LeEco Le S3 uses Helio X20 and has source code available,
Elephone s7 uses Helio X20 and has source code available
Only redmi note 4 MTK is missing, and I know it's NOT MediaTek's fault, Redmi Note 4 MTK users also deserve custom rom

Zobeta, 2017-12-09:
I remember that xiaomi never released any kernel sources of MTK devices.

Ayrtown-Karlos, 2017-12-09:
If Xiaomi really cares about her customers she should free

luk1337, 2017-12-09:
/me yawns

luca020400, 2017-12-09:
If Xiaomi really cares about her customers she should free

- Ayrtown-Karlos 2017 -

blingerpc, 2018-06-10:
LeEco Le S3 does not have source code

Ayrtown-Karlos, 2018-06-10:
Are you sure?
http://opensource.le.com

blingerpc, 2018-06-15:
That is for Snapdragon version which is x522 while the mtk version is x626 or x622