MiCode/Xiaomi_Kernel_OpenSource #804: Release kernel sources for Chiron

pdigennaro, 2017-12-11:
Please release kernel sources for the Chiron device (MIX 2).
GPL

luca020400, 2017-12-11:
Just check if the fucking kernel has been already released.
Hint: IT IS.
Damnit

aholstenson, 2017-12-11:
@luca020400 No need for swearing and being an asshole. If the sources are available it would be very helpful to provide a reference to where they can be found as there is no branch in this repository for `chiron`. That way requests for this device won't pop up again.

pdigennaro, 2017-12-11:
@luca020400 I can only see some pieces of code in the sagit branch regarding MIX 2, but they are only hints... there's no defconfig for chiron. Maybe you can point the full chiron kernel code to us?

TheStrix, 2017-12-12:
@pdigennaro It is full.. Use the same defconfig.