MiCode/Xiaomi_Kernel_OpenSource #805: Please release kernel for Xiaomi Mi Max 2 (oxygen)

DustPanic, 2017-12-16:
Many peoples want to install custom firmware!
Please release kernel for mi max 2.
Thanks!

toxavrn, 2017-12-16:
Please release kernel for Xiaomi Mi Max 2 (oxygen) +

jawaadsaeed, 2017-12-16:
Well then, fuck Xiaomi

ryan-andri, 2017-12-16:
don't put your hope into xiaomi, just bought their device and then sell it for lower price.

123oleg1977, 2017-12-17:
ryan-andri
Unfortunately. I did the same, I sold my device and my wife and daughter also left, but I will also look for another brand after the new year, I know for sure that I will not return to this firm, I'm very sorry that they have such an attitude towards their customers, It's time to gain a reputation, but you can lose it in a moment ...

luk1337, 2017-12-17:
My dog also left Xiaomi :^)

DrWssn, 2017-12-23:
Plz realase the kernal

rupansh, 2018-01-01:
Publish kernel source for 5a

luk1337, 2018-01-01:
can you stop?

DustPanic, 2018-01-03:
luk1337
No

DustPanic, 2018-01-08:
Thank you Xiaomi! Published kernel!!!

closed 2018-01-08 by DustPanic