MiCode/Xiaomi_Kernel_OpenSource #815: incompleted kernel redmi note prime gucci

ihateu7, 2018-01-07:
so many errors in redmi note prime kernel gucci
drivers are main problem please

and kernel is not completed

please recompile and release new kernel
take it serious and please repair it

luk1337, 2018-01-07:
no one cares, stop spamming this tracker.

ihateu7, 2018-01-07:
then  why u are replying just keep calm i know what to do and what not to do

luk1337, 2018-01-07:
Cause I keep getting emails cause of your stupid "issues".

ihateu7, 2018-01-07:
oh ur the stupid guy 
because all devices have so much development but once check redmi note prime dveelopment 
we have knowledge but the not working kernel all devices have atleast lollipop for my device is in kitkat please check that kernel errors

luk1337, 2018-01-07:
that's sad. too bad no one cares

ihateu7, 2018-01-07:
ANY SOLUTION TO REACH XIAOMI

ryan-andri, 2018-01-07:
oh come on, please dont again !