MiCode/Xiaomi_Kernel_OpenSource #825: Please Release Kernel Source For Redmi 4A (rolex)

muralivijay, 2018-01-19:
How much days We are for kernel source ? 1 year old device Redmi 4A stil Not released any Mrashmallow or nougat Kernel source .. but your release kernel source for mi max 2 faster . what a cheating level Xiaomi. are you planing to broke GPLV2 ?

krasCGQ, 2018-01-27:
Code-Review-2
Review-1

ryan-andri, 2018-01-27:
lul

dedejamaludin, 2018-01-31:
I believe @xiaomi will follow GPL rules. But why so long ago ? Its been one and half year but kernel source still not release yet. What the LOL ???

muralivijay, 2018-01-31:
@dedejamaludin beacuse they forcsing users keep stay with #shitmiui rom . now days miui rom not good for msm8917 chips. very laggy when, also poor ram managemant, according mi forum moderator says very redmi devices kernel source will be released after 1 Year. but actually redmi 4A lunched in nov 2016 . till they not relaesed

MacTavishOT, 2018-01-31:
Release kernel source for Redmi 4A rolex. Xiaomi shouldn't keep rolex users away from developing.

JNYCLVS, 2018-01-31:
Release the Kernel Source for Redmi 4A (Rolex), for further development and support for our device

abdullatifreal, 2018-01-31:
Please release kernel source for Redmi 4A! We need it immediately! More than 1 year passed since the release of this device but still no kernel source!

jnkcode, 2018-01-31:
already more than a year has not been released yet?

dedejamaludin, 2018-01-31:
@muralivijay : i gave my $100 dollars to buy this device, why @xiaomi force me to want more ???
Now i also gave my time at least one year without kernel source in my rolex, is there another f***ing shit ???

krasCGQ, 2018-01-31:
Because it's Xiaom~~i~~eme

sagwascript, 2018-01-31:
Is it takes time to publish kernel source ? Ok then give us a release date, we will wait for it. So that means you plan to publish it, but if you don't, it means you're never plan it to be published, and that's a problem.
So gives us the release date, a serious one, or if you have something to be discussed, we can discuss it here, be informative, we are good users :)

ryan-andri, 2018-01-31:
ahh come on -__- dont cry please :(

krasCGQ, 2018-01-31:
~~lol~~

imronturner, 2018-01-31:
Release kernel source for Redmi 4A rolex, for development our devices.

raphielscape, 2018-02-01:
K nice meme

> Everyday we rant to Xiaomeme about Rolex, we want ETA Sauce, Sauce when, but Xiaomeme will be Xiaomeme, Xiaomeme is memelord, coz Xiaomeme is meme

So summary : No ketchup, Just sauce, Raw Sauce

krasCGQ, 2018-02-01:
~~lmfao~~

sayan-tan, 2018-02-01:
Again the shit Xiaomi is violating gpl rules. Release kernel source for Rolex aka Redmi 4A now. Why you people don't understand that MIUI is not good for low RAM phones. We need further development.

Dilip1998, 2018-02-01:
Xiaomi should release their kernel source as soon as possible why they are doing I don't understand.

tejasurya03, 2018-02-01:
Release kernel sourcefor redmi 4a xiaomi.Plz allow redmi 4a users for custom rom development

rasyaadd, 2018-02-01:
Ctrl C + Ctrl V. Is that hard?

krasCGQ, 2018-02-01:
^ ROFLMFAO

tremores, 2018-02-01:
Xiaomi please stop violation of gpl rules and release kernels for Redmi 4a ASAP.

avirk, 2018-02-01:
Xiaomi is keeping away rolex and santoni users from develop by not giving kernel sources and continue to violate linux kernel gpl. Don't know if there is anything we can do more than else to report here. There must be a legal action from Google against these kind if companies if they don't release sources after a time of of releasing device.

muralivijay, 2018-02-01:
@krasCGQ then why xiaomi only released mi max 2 and m1 A1 faster ?. they stil ignoring redmi 4A/4X

aaamirsingh, 2018-02-01:
It's a humble request to @XiaomiIndia and @Xiaomi to release the kernel source for Redmi 4a . It's been 1.5 year passed and kernel source is not released yet.
You should better follow your own rules.
Think about releasing kernel source before releasing your device whose kernel is licensed under GPLv2.

jawaadsaeed, 2018-02-01:
Also, don't forget Prada. They forgetting it too

muralivijay, 2018-02-01:
@jawaadsaeed why u asking kernel source ? try to port any CAF kernel source for prada beacuse chipset msm8937 based . it can boot caf redmi 3s team already do this before releasing kernel source

muralivijay, 2018-02-01:
but MSM8917 to hard Not booting with caf source blank screen and no usb always .. @xiaomi please release kernel source for rolex . we need drivers for boot

AryanPatidar, 2018-02-02:
Release the kernel source for all Rolex users plz

We want kernel source for Redmi 4A Right now!
@xiaomi @XiaomiUSA
@XiaomiIndia @XiaomiIndia @Android @AndroidAuth @Google @Android 
#Redmi4A
#wewantkernelsource
#ReleaseKernelSources
#opensourcebreakerxiaomi #Gplv2violation

viktorszabo2002, 2018-02-02:
Please release it.

krasCGQ, 2018-02-02:
@muralivijay bcs Xiaomi only cares about some of their phones... 🤔

krasCGQ, 2018-02-02:
Also Mi A1 kernel source was released under demand of users

luk1337, 2018-02-02:
no

muralivijay, 2018-02-02:
@krasCGQ . I think xiaomi still not responding ..So iam going to register complaint to GNU xiaomi violating GPL. Let see

jawaadsaeed, 2018-02-02:
Xiaomi have been violating the GPL for years, they are criminals who don't care about international copyright law. Your github issue is insignificant to them, pissing in the wind. The only way to get corporate monsters to change is to hit them in their wallets i.e. research your products and don't buy from criminals. Xiaomi are as likely to listen to you after you've been conned by them as a fake dealer is gonna give someone their money back after conning them. I know, getting hit by a criminal is not a nice feeling, but you gotta learn from it and get a bit more cynical.

jawaadsaeed, 2018-02-02:
There's a reason you had to buy your phone from some shitty internet third party reseller or the criminal's own "official" website.

The same way there's a reason why the dealer is yelling out "hot bud" next to a crowded train station and selling you "ganja" in a black bag (that's really parsley) for 20 dollars.

krasCGQ, 2018-02-02:
¯\_(ツ)_/¯

dedejamaludin, 2018-02-04:
Is it right ? Xiaomi will release kernel source for some unreleased devices ?

krasCGQ, 2018-02-04:
...what @dedejamaludin
rolex has been exist for more than 14 months already. They have rights to fight for their rights.

Androinterest, 2018-02-04:
It's Been More than 1 year. You hasn't released kernel Source For Redmi 4a (Rolex) . You are Violating the GpLv2 Rule . 

Release Now .

Vessel09, 2018-02-04:
Release it please

bams0038, 2018-02-04:
I believe@XiaoMiwill follow GPL rules. But why so long ago ? Its been one year but kernel source still not release yet. What the LOL???

Nibi-git, 2018-02-04:
I already have neurosis due to the expectation of the kernel source :(

[image]

dedejamaludin, 2018-02-05:
Is there still no rrspons from xiaomi ? Very cruel...

muralivijay, 2018-02-05:
Xiaomi still not responding

krasCGQ, 2018-02-05:
Code-Review-2
Verified-1

tremores, 2018-02-05:
@xiaomi you are no. 1 violetor of @GPLv2 licence. Bu we will never give up...

akhilnarang, 2018-02-05:
LMFAO

brianalviano, 2018-02-05:
Please respond #gpl2violation #xiaomi #redmi4A #kernelsource

luk1337, 2018-02-05:
this isn't facebook, stop using these silly hashtags.

punyapal, 2018-02-05:
It's more than a year please release the kernel source for redmi 4a you are violating
GPL and if u do not release kernel source for redmi 4a then you have to face consequences.

luk1337, 2018-02-05:
too bad they won't face any.

krasCGQ, 2018-02-05:
It's China, sir. @punyapal

luk1337, 2018-02-05:
-9

krasCGQ, 2018-02-05:
-∞

muralivijay, 2018-02-05:
@bri stop spamming

bams0038, 2018-02-05:
Please @xiaomi release kernel source for Redmi 4a.. We wait..

luk1337, 2018-02-05:
no

Dzikri630, 2018-02-05:
Please release kernel for redmi 4a

sunnyraj84348, 2018-02-05:
Release source

[62 more replies]

closed 2018-03-26 by muralivijay