MiCode/Xiaomi_Kernel_OpenSource #843: NEED THE OREO SOURCE OF MIA1

aalakib07, 2018-02-26:
The phone was released about 6 months ago!!! but they didn't release the source code! How shame!!! 
They publish Nougat source in the time of Oreo!! LOL!!! 

Shame on you Mi! 
No More Mi Phones! 
Good Luck!

dragonpt, 2018-02-28:
Will be up on March

bgcngm, 2018-02-28:
@dragonpt what are you doing here? Long time no see!

muralivijay, 2018-02-28:
@dragonpt Xiaomi does not release . New android kernel source. They will only release base firmware version . Like if a Xiaomi device launched with marshmallow . They will only release marshmallow kernel  . Also look other Xiaomi device land, mido, Kenzo, markw stil having marshmallow source . But how they get Oreo source. You need rebase nougat to Oreo using CAF source . I hope u understand this 

Also my mi A1 Oreo boots with rebase source

dragonpt, 2018-02-28:
There will be further releases, has been an busy week to what concerns to XI open source 
( as you can see by yourself)

Things will more straight forward now, hopefully

@bgcngm nice to see you here, didn't knew you worked /have a Xiaomi ;)

bgcngm, 2018-02-28:
@dragonpt MTK is past. :) Hit me up on hangouts.

0pLuS0, 2018-03-08:
STOP SPAMMING PEOPLE AND READ! OR NEXT TIME YOU MIGHT BE BANNED!

https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/854

m-p-3, 2018-03-16:
It' out
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/65d05bc7ed777f448743a1d41a9e737858603bc6

dragonpt, 2018-03-16:
Yes of course its released ;)

Cheera

wbsdty331, 2018-06-27:
Mi A1 is Android One, Please contact Google, not Xiaomi

scafroglia93, 2018-06-30:
Ahahahahaha

scafroglia93, 2018-06-30:
Ciao MAMMA!

wbsdty331, 2018-07-09:
Let me say it again: 
> **Mi A1 is Android One**

Why are you still ask xiaomi to release their kernel source?
I've said as much as I'm prepared to say at this point.

scafroglia93, 2018-07-09:
Android One devices are not managed by Google but by the device manufacturer (xiaomi)

wbsdty331, 2018-07-09:
@scafroglia93
Yes, Manufacturer is Xiaomi, But Mi A1 not running MIUI but Pure Android, You just ask Xiaomi to release Kernel Source, But Can you really be certain it works?
By the way, Mi A1 bootloader unlock is different from other Xiaomi device.

scafroglia93, 2018-07-09:
What's the problem ?

The sources are always managed by the manufacturer, it has nothing to do with the system
If you buy cheap junk it's not my fault !

wbsdty331, 2018-07-09:
Do as you please. I use OnePlus 6, Redmi Note 5 and iPhone X.

I will never reply this issue.

muralivijay, 2018-07-09:
Lol close this topic kernel source already released. So many spams coming

ryan-andri, 2018-07-09:
@scafroglia93 agreed.

dragonpt, 2018-07-09:
Sorry but you all seem just spoiled kids.

If XI doesn't release kernel, they are our archenemies
If they do, and just it happens to be a bit old, archenemies they are still

This ain't XDA or something

Does the released kernel source boots, and all the hardware is functional? Yes? Then?

As long as the released source match's the release , that's fine, and that's what is expected .

Now don't expected XI , or as matter of fact any OEM,to keep updating their public source on daily basis.

Just be thankful of what you have

scafroglia93, 2018-07-09:
Then explain to me why the phone manufacturer I'm currently using updates the CAF releases ?

dragonpt, 2018-07-09:
How many OTAs on the A1, has been released?

The kernel source released, match's the PR
, if there's any OTA afterwards with kernel changes, then yes, kernel source should be updated to match that release

YaroST12, 2018-07-09:
@scafroglia93 probably because its userbase is 1/1000 of Xiaomi? If Xiaomi releases a bugged update - it's getting mentioned on all Android news sites. If BQ releases a bugged update... Doubt anyone will even notice.