MiCode/Xiaomi_Kernel_OpenSource #849: 请愿之开放红米4A内核源码

yougar0, 2018-03-05:
我在下面看到外国好多人都在请愿开放内核源码，确实，小米应该开源了。国内也有好多人需要红米4A内核源码，比如说我。最近刚买了红米4A，为了做实验，必须得到完整内核代码，所以，希望维护小米开放源码组织的看到，能尽快开源此版本代码。

maxismangh, 2018-03-07:
Exactly. What is Xiaomi doing with that kernel source...

ivan19871002, 2018-04-09:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/riva-n-oss

Kernel: Xiaomi kernel changes for Redmi 5, Redmi 4A, Redmi 5A.

The Patch based on QualComm release version: LA.UM.5.6.r1-05900-89xx.0
The kernel config file used are:
rosy-perf_defconfig(Redmi 5), S88505AA1-perf_defconfig(Redmi 5A), S88503AC1-perf_defconfig(Redmi 4A)

closed 2018-04-09 by ivan19871002

wangcong15, 2020-03-07:
> https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/riva-n-oss
>
> Kernel: Xiaomi kernel changes for Redmi 5, Redmi 4A, Redmi 5A.
>
> The Patch based on QualComm release version: [...]

Thanks for you reply. 
Could you tell me whether this kernel is compatible with the development version? I have a Redmi 4A (in development version ROM). I wanna flash the `boot.img` to update the kernel image and keep the ROOT. But my phone fails to boot with the kernel image built from here.