MiCode/Xiaomi_Kernel_OpenSource #852: how to change the boot.img with MIX2

zhr1991, 2018-03-07:
I use the sagit-o-oss branch, after compile the kernel code( change the code with a patch in oneplus5 kernel), using toolchain ndk-r15c, sagit_user_defconfig

then, I got the boot.img from miui8.3.1.zip, using xiaolu(github)'s mkbooting tool, I change the kernel to arch/arm64/boot/Image.gz-dtb, and repack to the img

after I flash the img to my mix2, it will go to mi unlocked screen and won't boot up. I use the Image,Image.gz also, it will loop reboot also.

So, can somebody help me? what should I do to compile and flash my kernel that can work

muralivijay, 2018-03-08:
First backup your current rom boot image . Spilt it using android image Kitchen. Copy kernel compiled kernel zimage from arch/(your arch)/boot/image.gz . Copy it to spilt_img folder . And remove the old kernel repack it . Then try to boot by via fastboot for testing "fastboot boot image-new.img".

zhr1991, 2018-05-17:
@GeneBlue , Sorry ,I don't know how to solve that. I have little knowledge on kernel build...

GeneBlue, 2018-05-18:
@zhr1991  Hi, you can use AIK-Linux-v3.2-ALL tool to repack Mix2 boot.img, you can download it from here https://forum.xda-developers.com/showthread.php?t=2073775

Do not root your device before reflash boot.img

- use AIK-Linux unpack original boot.img 
- replace zImage in split_img folder with you compile kernel
- edit default.prop file in ramdisk, delete "ro.secureboot.devicelock=1" line
- edit fstab.qcom file in ramdisk,  remove ", verfity" in line "/dev/block/bootdevice/by-name/system"
make "encryptable=footer" in line "/dev/block/bootdevice/by-name/userdata"
- edit init.miui.rc file, delete
```
# secureboot
service setlockstate /sbin/setlockstate
    class core
    oneshot
    seclabel u:r:vold:s0
```
My device  trapped in boot loop when repacking and reflashing my new_boot.img。 I think the cause is my device was rooted by using xiaomi root tool in develop rom. I did not try again. Maybe you can try upper suggestion without rooting your device. If you success, please leave you message and at me.

zhaotianxing, 2018-06-21:
@GeneBlue it does not help...