MiCode/Xiaomi_Kernel_OpenSource #856: add sources for redmi note 2(hermes)

nofearnohappy, 2018-03-10:
MTK rip

Smosia, 2018-03-21:
RIP

DumF0rGaming, 2018-03-21:
F
[image]

mario-rr, 2018-03-21:
Rest in Pepperoni

andrea2107, 2018-03-21:
A "long cheer" for the release of the sources!

Rip.

joe2k01, 2018-03-21:
[image]

MZO9400, 2018-03-21:
RIP²

PsyMan47, 2018-03-21:
Pls release kernel for mi 7

nofearnohappy, 2018-03-21:
-
[image]

nofearnohappy, 2018-03-21:
[image]

nofearnohappy, 2018-03-21:
[image]

GiaSen, 2018-03-21:
Release MIUI oreo for mido

krasCGQ, 2018-03-22:
Wrong thread @GiaSen

muralivijay, 2018-03-22:
RIP MTK

fulgerul46, 2018-03-22:
rip

jawaadsaeed, 2018-03-26:
RIP MediaKetek

nofearnohappy, 2018-04-14:
+

DoveKi, 2018-04-17:
Mi Pad 3 also MTK, but Xiaomi open the source code now.

dragonpt, 2019-01-17:
It's released, have fun

closed 2019-01-18 by ivan19871002