MiCode/Xiaomi_Kernel_OpenSource #861: Xiaomi Redmi 5 Plus (vince)

obsvg216, 2018-03-26:
Please post the kernel source code. We are all very much waiting for this.

mdeejay, 2018-04-03:
Xiaomi WTF? Where source codes for vince?

mohancm, 2018-04-03:
@mdeejay you got Vince?

mdeejay, 2018-04-03:
yup

mdeejay, 2018-04-03:
bombing Xiaomi twitter right now.

mohancm, 2018-04-03:
Lol

mdeejay, 2018-04-08:
And what you waiting, Xiaomi?
Stop violating GPL! Release all opensource for vince!

luk1337, 2018-04-08:
cry more, it's definitely going to help.

Blacksuan19, 2018-04-08:
yo bitch ass niggas released source for whyred and not vince wtf

InZaneDanii, 2018-04-08:
Please Release the Kernel Source for Redmi 5 Plusby modifying and distributing GPL software, this has to comply with the GPL terms, in this case GPLv2. And therefore this I kindly request you release kernel sources otherwise this would be copyright infringement.According to the GPL v2 license terms all modifications made to the Linux kernel must be released to the public. Please follow the terms of the license you are using. Thank you

TeamCampurAduk, 2018-04-08:
Whats wrong with you Xiaomi??! You release source code for whyred, but for Vince cant??? WTF!!

AakashPandey, 2018-04-09:
Release the source your 9.5 is bloated with spywares, we need to jump to aosp / lineage

closed 2018-04-13 by obsvg216

thenanobel, 2018-04-25:
@obsvg216 Why have you closed the issue? Have them released the source code?

obsvg216, 2018-04-25:
@thenanobel yes, in tiffany-n-oss branch