MiCode/Xiaomi_Kernel_OpenSource #866: wtf xiaomi

nofearnohappy, 2018-03-31:
You have opened the source code for MTK for XiaoMi Pad3.
Do not mock me open Redmi Note 2(hermes)

luk1337, 2018-03-31:
lol.

rakusakura, 2018-04-01:
And Xiaomi 5X

ivan19871002, 2018-04-09:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cappu-n-oss

Are you sure you've read the submissions?

nofearnohappy, 2018-04-09:
>
>https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cappu-n-oss
>Are you sure you've read the submissions?

course studied and downloaded the source
I myself collect a custom kernel for my redmi note 2( https://github.com/nofearnohappy/kernel_6.0_hermes )
I have problems with only cw2015 (battery drivers), consumerir(IR BLASTER), bmi160 and mag sensors
So it goes
For this reason I studied the source code for  cappu-n-oss   trying to find useful drivers

ibilux, 2018-07-11:
open the source code

n00bundercover, 2018-07-11:
Releasing this source would restore my trust in xiaomi.

scafroglia93, 2018-07-16:
Ciao MAMMA