MiCode/Xiaomi_Kernel_OpenSource #880: mt6795 open the source code(hermes)

nofearnohappy, 2018-04-16:
open the source code

Golbinex, 2018-04-16:
No source codes for MTK for some reason, maybe they don't even have them anymore. Just get new phone, all qualcomm phones have sources now, I think

mdeejay, 2018-04-16:
MTK sucks and should die

scafroglia93, 2018-04-17:
Come to BQ if you want full mtk kernel source

DoveKi, 2018-04-17:
@Golbinex Mi Pad 3 also MTK, but it's already open source.
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/cappu-n-oss

Golbinex, 2018-04-18:
Mi pad 3 is new device, old devices will probably never have sources for some reason, otherwise they would release it already.

yuvraj99, 2018-04-20:
If xiaomi have released source code for one mediatek device then they have to release it for all according to gpl if they have posted it for one device not for others you have right to ask for sources and if they not provide you!, they'll simply loose the right to sell their smartphones and loose the right to use Linux kernel!
Clear that to them!

Dinolek, 2018-05-03:
@ivan19871002 Can you say us at least, why you can't share it?

ibilux, 2018-07-11:
open the source code

diarf, 2018-07-11:
Open the source code please

scafroglia93, 2018-07-11:
Buy another phone man

diarf, 2018-07-11:
@scafroglia93 if u want to buy for me you're welcome

ohjames, 2018-08-21:
@JaEv2 you don't understand the GPL. Xiaomi are already infringing the GPL whether they release the source code for one device, zero devices or multiple devices.

> If xiaomi have released source code for one mediatek device then they have to release it for all according to gpl