MiCode/Xiaomi_Kernel_OpenSource #898: Please release kernel source for suspect Longcheer ODM phones(R4,RN5A)

copslock, 2018-06-02:
Xiaomi had released kernel sources of most of it's devices which ODM by Wingtech or Huaqin,like popular Redmi Note 4(X) S625 version,Redmi 5 Plus, Redmi 4 Prime, Redmi 4A/5A ,Redmi 4X, 3S/S,all of these phones are almost ODM by W$H,but what about the suspect Longcheer ODM models?It's weired that Redmi Note 3(kenzo),Redmi note 5(whyred) all ODM by Longcheer.Anyone can give some explain?

nervehammer, 2018-06-05:
well @copslock ugg/ugglite source is here now. You can check this https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/ulysse-n-oss

copslock, 2018-06-07:
@nervehammer Good,but still absence of the Redmi 4 2016090 msm8937 model codename prada

wbsdty331, 2018-07-09:
FIH?
I think you can google how to unlock Nokia's Android Phone Bootloader. And Try it for your FIH Redmi 4

Kizoky, 2018-08-15:
@nervehammer 
That's Redmi Note 5A, not Redmi 4 prada