MiCode/Xiaomi_Kernel_OpenSource #899: Kernel doesn't boot at all(Redmi Note 5) vince

gxr441, 2018-06-04:
Compiled kernel with vince-perf_defconfig, encountered lot of header not found errors and many forbidden warnings(should I have disabled these?), When booting with fastboot, fastboot says booting FAILED, and just black screen, i have to force it to reboot back. What's the problem?

copslock, 2018-06-07:
The codebase dosen't provide enough resources like Device Tree,you may need more research on arm64 boot.img structure

gxr441, 2018-06-07:
I think there is dtb and I didn't create a boot image, I was trying to boot by providing fastboot with kernel and default ramdisk.

TingyiChen, 2018-07-26:
source no issue.

gxr441, 2018-07-26:
No issue with source, but I don't know what went wrong here, has anybody else got tons of header not file errors and warnings?

TingyiChen, 2018-07-26:
compile with rom inline. it will boot

gxr441, 2018-07-26:
That's exactly what I am trying now!:). syncing android code base is ...

cnjhb, 2019-12-10:
Brother, did you solve this problem?

Morrigan-Ship, 2025-05-24:
same forbidden warning errors