MiCode/Xiaomi_Kernel_OpenSource #904: mix2s开源残废，你们能跑起来？看看隔壁等于7！

sekaiacg, 2018-06-16:
miui上跑不了，不明白开源有什么用，自欺欺人，就像完成任务一样
mi5也一样，老外说没更新代码，不支持miui。

ivan19871002, 2018-06-20:
不能开机我能开源出来？

ivan19871002, 2018-06-20:
我看看最新版本能不能起来吧。开源的那天是可以开机的 并且做过功能验证的，必须保证功能正常才能开机的。

ivan19871002, 2018-06-20:
为了保证一切正常，我从github 把代码拉到本地编译的。
测试一切正常，开机没有任何问题，功能也测试了一下，也正常。

closed 2018-06-20 by ivan19871002

simonsmh, 2018-06-24:
@ivan19871002 秋大好，我也是一位mix2s用户，使用的是国内6+128g非全网通的xe版的机子，刷了global的rom，就没办法使用源码编译的kernel开机，而国内版的rom正常。这与刷入magisk时对原kernel进行patch后无法开机的表现是相似的。请问我应该如何处理相关的错误？感谢！

ivan19871002, 2018-06-26:
@simonsmh  你最好是重新开个issue，并且提供log 我才能帮你解决问题

simonsmh, 2018-06-26:
@ivan19871002 不是啊，MIX2S有全球频段的版本m1803d5xa和非全球频段的版本m1803d5xe，我手上的是m1803d5xe版本。
我不知道怎么提供log，因为没法获取log（

simonsmh, 2018-06-26:
@ivan19871002 MIX2s分全球频段m1803d5xa的国内8g+256g内存版本和非全球频段m1803d5xe的国内6g内存版本，我手上是非全球频段的版本。这个xe版本甚至刷global之后lock bootloader都会出现bootloop，很怪（
而且log不知道怎么提，因为根本没法开机的

ivan19871002, 2018-06-27:
没有log 我就很难办了啊。重新开个issue讨论这个问题吧。不要到这里讨论了

sanqiuu, 2023-12-05:
> 没有log 我就很难办了啊。重新开个issue讨论这个问题吧。不要到这里讨论了

5.4.86 civi1s你能开机就能见了鬼了