MiCode/Xiaomi_Kernel_OpenSource #910: mix2s内核在miui10上 蓝牙无法使用，求解

sekaiacg, 2018-06-26:
06-26 13:53:11.270  1002  6340  6359 D bt_device_profile: Entry is already present in the list
06-26 13:53:11.270  1002  6340  6359 W bt_device_profile:  PBAP_ID: key :: pbap_0102_support, value :: true
06-26 13:53:11.270  1002  6340  6359 W bt_device_profile:  PBAP_ID: key :: use_sim_support, value :: true
06-26 13:53:11.270  1002  6340  6359 D bt_device_profile: Entry is already present in the list
06-26 13:53:11.270  1002  6340  6359 W bt_device_profile:  MAP_ID: key :: map_email_support, value :: true
06-26 13:53:11.270  1002  6340  6359 I bt_stack_config: init attempt to load stack conf from /etc/bluetooth/bt_stack.conf
06-26 13:53:11.270  1002  6340  6364 I bt_osi_thread: run_thread: thread id 6364, thread name bt_jni_workqueue started
06-26 13:53:11.270  1002  6340  6359 I bt_stack_manager: event_init_stack finished
06-26 13:53:11.270  1002  6340  6340 I bt_btif : get_profile_interface: id = socket
06-26 13:53:11.270  1002  6340  6364 E bt_btif_storage: btif_storage_get_adapter_property: Controller not ready! Unable to return Bluetooth Address
06-26 13:53:11.270  1002  6340  6364 E BluetoothServiceJni: adapter_properties_callback: Status 1 is incorrect
06-26 13:53:11.271  1002  6340  6340 I bt_btif : get_profile_interface: id = sdp
06-26 13:53:11.271  1000  1567  1567 D BluetoothManagerService: Bluetooth Adapter name changed to QCOM-BTD
06-26 13:53:11.272  1000  1567  1567 D BluetoothManagerService: Stored Bluetooth name: QCOM-BTD
06-26 13:53:11.272  1002  6340  6364 D BluetoothAdapterProperties: Name is: QCOM-BTD
06-26 13:53:11.273  1002  6340  6340 I BluetoothAdapterService: Phone policy enabled
06-26 13:53:11.273  1002  6340  6340 D BluetoothBondStateMachine: make
06-26 13:53:11.274  1002  6340  6340 D BluetoothAdapterService: setAdapterService() - set to: null
06-26 13:53:11.274  1002  6340  6365 I BluetoothBondStateMachine: StableState(): Entering Off State

ivan19871002, 2018-06-26:
我查一下

ivan19871002, 2018-06-26:
查了一下，最新高通推了一个新的patch，上层更新了导致底层有一个地方不匹配。 已经更新，地址：https://github.com/MiCode/Xiaomi_Kernel_OpenSource/commit/6422c397b09826f8e1f9ab75a53e25d472b488cd

closed 2018-06-26 by ivan19871002