MiCode/Xiaomi_Kernel_OpenSource #923: 你好，小米code，libra-n-oss的wifi貌似存在致命错误。

zah0, 2018-08-03:
和我预期想象不一致，wifi conflict
望修复！！！
[image]
wifi conflict
wifi conflict
wifi conflict
wifi conflict
wifi conflict
wifi conflict

zah0, 2018-08-03:
编译器为gcc 4.9
`make libra_user_defconfig`
`make -j8`
`08-03 20:34:39.093 +0000  2310  2507 E WifiStateMachine: ### **_Failed to load driver_**
08-03 20:34:39.096 +0000  2310  2310 E WifiController: WifiControllerWifi turn on failed
08-03 20:34:39.100 +0000  2570  2570 D ToggleManager: MiDrop: updateMiDropToggle(boolean) isMiDropOn = false isWifiApDisabled = false mMiDropChanging = false mWifiChanging = false
08-03 20:34:39.100 +0000  6670  6670 D LiteToggleManager: mWifiEnable = false mWifiChanging = false wifiState = 4
08-03 20:34:39.100 +0000  6670  6670 D LiteToggleManager: isWifiEnabled=false;isWifiContected=false;intent=android.net.wifi.WIFI_STATE_CHANGED
08-03 20:34:39.100 +0000  2570  2570 D ToggleManager: updateWifiToggle wifiState=4 mWifiConnected=false action=android.net.wifi.WIFI_STATE_CHANGED
08-03 20:34:39.102 +0000  6670  6670 D TouchAssistant: updateIcon entry:lock_screen
08-03 20:34:39.104 +0000  6670  6670 D TouchAssistant: updateIcon entry:screenshot
08-03 20:34:39.105 +0000  6670  6670 D TouchAssistant: updateIcon entry:key_press_menu
08-03 20:34:39.106 +0000  6670  6670 D TouchAssistant: updateIcon entry:key_press_home
08-03 20:34:39.106 +0000  6670  6670 D TouchAssistant: updateIcon entry:key_press_back`

zah0, 2018-08-03:
08-03 20:34:39.093 +0000  2310  2507 E WifiStateMachine: Failed to load driver
08-03 20:34:39.096 +0000  2310  2310 E WifiController: WifiControllerWifi turn on failed
08-03 20:34:39.100 +0000  2570  2570 D ToggleManager: MiDrop: updateMiDropToggle(boolean) isMiDropOn = false isWifiApDisabled = false mMiDropChanging = false mWifiChanging = false
08-03 20:34:39.100 +0000  6670  6670 D LiteToggleManager: mWifiEnable = false mWifiChanging = false wifiState = 4
08-03 20:34:39.100 +0000  6670  6670 D LiteToggleManager: isWifiEnabled=false;isWifiContected=false;intent=android.net.wifi.WIFI_STATE_CHANGED
08-03 20:34:39.100 +0000  2570  2570 D ToggleManager: updateWifiToggle wifiState=4 mWifiConnected=false action=android.net.wifi.WIFI_STATE_CHANGED
08-03 20:34:39.102 +0000  6670  6670 D TouchAssistant: updateIcon entry:lock_screen
08-03 20:34:39.104 +0000  6670  6670 D TouchAssistant: updateIcon entry:screenshot
08-03 20:34:39.105 +0000  6670  6670 D TouchAssistant: updateIcon entry:key_press_menu
08-03 20:34:39.106 +0000  6670  6670 D TouchAssistant: updateIcon entry:key_press_home
08-03 20:34:39.106 +0000  6670  6670 D TouchAssistant: updateIcon entry:key_press_back

GuaiYiHu, 2018-08-17:
WiFi是需要自己加入qcacld-2.0的module一起编译的... 和内核是分离的...

zah0, 2018-08-30:
@GuaiYiHu 你好，乖奕虎。我这儿有个问题。

使用的是 > quic-la-platform-vendor-qcom-opensource-wlan-qcacld-2.0-LA.BF64.1.2.3_rb1.7
这个版本的qcacld-2.0

```
In file included from drivers/staging/qcacld-2.0/CORE/SERVICES/COMMON/adf/adf_os_mem.h:40:0,
                 from drivers/staging/qcacld-2.0/CORE/SERVICES/COMMON/adf/linux/adf_os_dma_pvt.h:33,
                 from drivers/staging/qcacld-2.0/CORE/SERVICES/COMMON/adf/adf_os_dma.h:39,
                 from drivers/staging/qcacld-2.0/CORE/SERVICES/COMMON/adf/adf_nbuf.h:43,
                 from drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_packet.h:60,
                 from drivers/staging/qcacld-2.0/CORE/VOSS/inc/i_vos_list.h:46,
                 from drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_list.h:73,
                 from drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_api.h:64,
                 from drivers/staging/qcacld-2.0/CORE/WDA/inc/legacy/palTypes.h:46,
                 from drivers/staging/qcacld-2.0/CORE/WDA/inc/legacy/halTypes.h:45,
                 from drivers/staging/qcacld-2.0/CORE/SME/inc/ccmApi.h:43,
                 from drivers/staging/qcacld-2.0/CORE/SME/inc/sme_Api.h:46,
                 from drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.c:49:
drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_cnss.h: In function 'vos_oob_enabled':
drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_cnss.h:559:2: error: implicit declaration of function 'cnss_wlan_query_oob_status' [-Werror=implicit-function-declaration]
  if (-ENOSYS == cnss_wlan_query_oob_status())
  ^
drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_cnss.h: At top level:
drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_cnss.h:565:48: error: unknown type name 'oob_irq_handler_t'
 static inline int vos_register_oob_irq_handler(oob_irq_handler_t handler,
                                                ^
drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_cnss.h: In function 'vos_unregister_oob_irq_handler':
drivers/staging/qcacld-2.0/CORE/VOSS/inc/vos_cnss.h:573:2: error: implicit declaration of function 'cnss_wlan_unregister_oob_irq_handler' [-Werror=implicit-function-declaration]
  return cnss_wlan_unregister_oob_irq_handler(pm_oob);
scripts/Makefile.build:308: recipe for target 'drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.o' failed
make[3]: *** [drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.o] Error 1
scripts/Makefile.build:455: recipe for target 'drivers/staging/qcacld-2.0' failed
make[2]: *** [drivers/staging/qcacld-2.0] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/staging' failed
make[1]: *** [drivers/staging] Error 2
make[1]: *** 正在等待未完成的任务....
Makefile:819: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```
查看了这里所有的头文件，没有发现问题
make -j16跳不过去，还停下来了

GuaiYiHu, 2018-08-31:
https://github.com/xiaomi-note-dev/android_kernel_xiaomi_leo/commits/mko-mr1/drivers/staging/qcacld-2.0

zah0, 2018-08-31:
你好，你的android_kernel_xiaomi_leo问题依旧重现。

但是我去includes/linux替换了一些文件后修复了以上问题，
但是又出现一个新的问题。
```
CC drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.o
drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.c:50:31: fatal error: wlan_hdd_includes.h: No such file or directory
#include "wlan_hdd_includes.h"
^
compilation terminated.
scripts/Makefile.build:308: recipe for target 'drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.o' failed
make[3]: *** [drivers/staging/qcacld-2.0/CORE/HDD/src/wlan_hdd_assoc.o] Error 1
scripts/Makefile.build:455: recipe for target 'drivers/staging/qcacld-2.0' failed
make[2]: *** [drivers/staging/qcacld-2.0] Error 2
scripts/Makefile.build:455: recipe for target 'drivers/staging' failed
make[1]: *** [drivers/staging] Error 2
Makefile:819: recipe for target 'drivers' failed
make: *** [drivers] Error 2
```
硬问题，解决不了
我觉得是include/linux/问题，但是我不太会码C 。

GuaiYiHu, 2018-09-02:
您还是... 弃坑吧...

Gah0, 2018-09-07:
已解决问题
add in qcacld-2.0 kbuild
`EXTRA_CFLAGS += -Wno-implicit-function-declaration -Wno-int-conversion`

closed 2018-09-07 by Gah0