MiCode/Xiaomi_Kernel_OpenSource #925: Please release the source code for Mi Pad 4

arter97, 2018-08-05:
It's nearly reaching 2 months since the release.

Please release the kernel source for the Mi Pad 4.

PythonLimited, 2018-08-09:
I really hope that they will release it in under 3 months after release as they told...

ivan19871002, 2018-08-12:
We will release it this month.

PythonLimited, 2018-08-12:
thanks for the reply

PythonLimited, 2018-08-22:
Dunno, seems like they force us to wait

PythonLimited, 2018-08-25:
Amy update @ivan19871002 ? Its only 6 days legt this month

arter97, 2018-08-28:
So the Mi A2 kernel sources were released today, and we're still yet to see Mi Pad 4's source.

@ivan19871002 @liuguo09 You guys have 3 days left. If the sources are not released til that point, you can count on me to get media coverage on how Xiaomi is ignoring yet another legal obligation and breaking a promise with the community.

Get your shit together.

PythonLimited, 2018-08-28:
Count me in, in the last two days two more sdm660 kernels have been released but no sign or update about the Mi Pad 4.
Just frustrating

Hero2457, 2018-08-29:
Legit just created an account here just to bash Xiaomi. 

Ya'll better get the kernel source out for the Mi Pad 4.

PythonLimited, 2018-08-29:
Even the new pocophone got one. Why is the mi pad 4 so late??

rcstar6696, 2018-08-30:
Waiting too now

PythonLimited, 2018-08-31:
Last day if we can trust the words of @ivan19871002 ....

PythonLimited, 2018-09-02:
Hello, @ivan19871002 we had the same situation nearly a month ago. JUST RELEASE IT. U SHOULD EVEN IMMIEDIATELY RELEASE THE SOURE CODE AFTER LAUNCHING THE DEVICE, IF NOT ITS VIOLATING THE GPL!!!! And relly telling everyone we will release it 3 months after release is a shame for a company like u. Take a look at Samsung they give u the links 1 to 2 days after. And thats also something why u will never grow that much as company if u dont get this working. I mean i wont fill a lawsuit against u, but if some companies decide to u are f*****. There are still devices from one year ago which havent recieved their kernel source. Fix that Xiaomi and u will See that this part oft the community will be happy.

rcstar6696, 2018-09-03:
Up! We are still waiting

PythonLimited, 2018-09-03:
The display of my pad started malfunctioning today. Let's see if the kernel got released when I get my replacement. :D

furlonb3, 2018-09-03:
Any updates on this?

BartekM01, 2018-09-05:
Dear @ivan19871002, when We can expect release of MiPad4 source code? 
Thank You

BartekM01, 2018-09-08:
@liuguo09 @ivan19871002 we have September and still there is no sources released. Any news about release?

Trozmagon, 2018-09-11:
This is very unfortunate, I bought this device because I though xiaomi were pretty honest with regards to releasing sources, guess I was wrong.

BartekM01, 2018-09-18:
Hi again...
@liuguo09, @ivan19871002. How long it's gonna take? Xiaomi is VIOLATING the GPL!!! Please release sources ASAP!

PythonLimited, 2018-09-20:
Ok I got my repaired pad back, but where the kernel?? 
It's even getting close to 3 months in which u told you'll release the kernel.

PythonLimited, 2018-09-20:
And the only reason I can think of is that they either have some source code they don't want to share in the kernel an be to outsource first, or they use some proprietary code in like the camera or wherever else. But I mean they should have cleared the first with the OEMs or didnt released the pad until they have...

zvanderbilt, 2018-09-26:
I would also like to see the kernel source released.

moham96, 2018-09-30:
we need the kernel source for MI Pad 4 and MI Pad 4 plus

PythonLimited, 2018-10-05:
Two months since this issue has been opened....😐😥😫

SpaceAgeHero, 2018-10-08:
Let's start a Shitstorm on Twitter.

PythonLimited, 2018-10-08:
jo im in

PythonLimited, 2018-10-12:
A friend of mine just reported this to GNU :D.
But I hardly assume it will do nothing.
Or someone from The Linux kernel contributor team could enforce this... Hopefully

PythonLimited, 2018-10-15:
What we could try in meantime is building a kernel with whyred base and our config.
Unfortunately I don't got time for such a time consuming project.
Also it seems that most drivers for display and touch are missing...
Here's the .config link
https://mega.nz/#!ad8A1IiJ!jLqoW5QaoR29BNJTBTvcjMdHLAlP649cDWKGeb9AHeg

zvanderbilt, 2018-10-15:
I understand your eagerness to roll your solution to this but trying to build a kernel for a tablet that is targeted towards a phone might end up sending you down a rabbit hole. I would definitely recommend not using a canned build.config either. Just take the time and go through the kernel menuconfig and add what you need and remove what you don't. Once/if they release the kernel I'll definitely post whatever I make to share.

PythonLimited, 2018-10-16:
Well this config IS from Xiaomi :p 
U can extract the running kernel config from /proc if the kernel is configured to provide support for that.

arter97, 2018-10-16:
Unless you're willing to reverse engineer display and touch protocol, that's a near impossible task.
Not even worth starting it.
Device tree also differs a lot.

PythonLimited, 2018-10-16:
Even XDA writes about that now.
Cmon Xiaomi
https://www.xda-developers.com/xiaomi-mi-pad-4-kernel-source-code-not-released/

PythonLimited, 2018-10-20:
Not only for u :(

zyonbao, 2018-10-25:
Any progress on this issue?

ArturPugacescu, 2018-10-30:
Disappointed, Oneplus 6t got their kernel source code released next day after the device was revealed. https://github.com/OnePlusOSS/android_kernel_oneplus_sdm845/commit/ea0847d32a41c41320333113966f8a719d482f47

sxxylz, 2018-10-31:
I'm still waiting...
any progress?

PythonLimited, 2018-10-31:
Nah :/

steadramon, 2018-11-07:
Weekly prod....

Are Xiaomi actually planning to release this Kernel? @ivan19871002 @liuguo09

thomasvt1, 2018-11-07:
> Weekly prod....
> 
> Are Xiaomi actually planning to release this Kernel? @ivan19871002 @liuguo09

I think it will be a long wait...

steadramon, 2018-11-07:
https://en.miui.com/forum.php?mod=viewthread&tid=4025675

PythonLimited, 2018-11-07:
well i asked there too a month ago or so
no answer but even the 6 months are nearly over...

zvanderbilt, 2018-11-07:
The best thing we can do at this point is just think back to this issue when we want to purchase a new device to develop on. 

If oneplus can release their source code the day after a release there is no reason xiaomi shouldnt do the same, if nothing more than to compete.

filjusz, 2018-11-09:
What a fu*kin shame! I will never buy xiaomi device ever again

PythonLimited, 2018-11-09:
I would but only after the kernel release...

Mirr0ch1, 2018-11-15:
nearly the end of 2018, still no sign of code-releasing.

Deses, 2018-11-16:
Still waiting for kernel sources to buy this tablet. I'll NOT use MIUI, not even once.

moham96, 2018-11-16:
I'm waiting the sources to buy the tablet as well

PythonLimited, 2018-11-16:
Maybe a Christmas present for us XD?

filjusz, 2018-11-17:
I dont think they will relase the sources...

Mirr0ch1, 2018-11-23:
yeah, been so long

arter97, 2018-11-23:
I was literally hours away from selling my Mi Pad 4.

Thanks but no thanks. You are still a disgrace for the community and I won't purchase another Xiaomi products just because of this horrible experience.

Fuck you.

closed 2018-11-23 by arter97