MiCode/Xiaomi_Kernel_OpenSource #926: kernel source Ulysse

Aarqw12, 2018-08-09:
Hello  oreo for redmi note 5a/prime planned? And will be oreo kernel sources available please ? Is necessary ... We already have oreo custom rom, but there is not camera and flashlight working. Could you pleases release oreo kernel sources for redmi note 5a/prime, or at least help fixing camera and flashlight?

thanks for your response

DarkLordLPCZ, 2018-08-09:
Yes please

DarkLordLPCZ, 2018-08-09:
We are waiting so long xiaomi

DarkLordLPCZ, 2018-08-09:
Xiaomi please release oreo kernel sources so we can fix camera!!!!

Haise1991, 2018-08-09:
Release Oreo kernel sources, please. 🖤

nayefhaidir, 2018-08-09:
Yes Release THAT!!!

Hryniuk1, 2018-08-09:
better to skip oreo and release pie

stydent21rus, 2018-08-09:
pliz oreo fix cam !!!!

reza740, 2018-08-10:
Please give us freedom for thats reason Xioami,thanks...

adislice, 2018-08-10:
Please xiaomi....

closed 2018-08-12 by ivan19871002

Aarqw12, 2018-08-13:
thanks to close without response ...

luk1337, 2018-08-13:
@Aarqw12 you're welcome!

reza740, 2018-08-13:
Why this clossed?I dont get it...

DarkLordLPCZ, 2018-08-13:
@reza740 same.... They don't give a fck about us

DarkLordLPCZ, 2018-08-14:
K