MiCode/Xiaomi_Kernel_OpenSource #927: xiaomi mi8

1nfrag, 2018-08-14:
I know this isnt the mi8 project but if you could provide an ETA for xiaomi mi 8 kernel source that would be greatly appreciated!

Thanks!

1nfrag, 2018-08-16:
pretty please?

1nfrag, 2018-08-21:
Not trying to be rude but I'd like some clarification to when/if Xiaomi will release the source for Mi8? Could you help out with this please @ivan19871002 ?
Sorry for tagging you but you seem to consistently reply with any knowledge so if you could help? Thanks!

ZeNiXxX, 2018-08-23:
Here too waiting for kernel Release for Dipper. I cant build Havoc OS, Aosp Pie Roms and so on for Dipper.

Please release the sources

bassbounce, 2018-08-23:
Waiting for it too, can u give more information about the release date for the source?

kalon33, 2018-08-23:
Duplicate of https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/922.
In this original issue, it has been declared "It is expected to be open source in August, and there have been a lot of things lately. The process has not finished"

viabary, 2018-08-23:
Anxiously waiting for a release, to begin dev of custom ROMs. Thank you Xiaomi in advance!

parala, 2018-08-24:
@1nfrag I am a Chinese developer, I'm axiously waiting for the source code as well. I'm pretty sure that xiaomi will eventually release the code. According to the developer's words, the code should have been released in August.But there is still some work to do.so they didnt release it.I think it's mainly because they're working hard to get Android P working on dipper. This is undoubtedly a groundswell in Chinese forum:)

carolef1984, 2018-08-25:
1nfrag, you did some great work for the x2. I have now got mi8 and am eagerly awaiting the release of the code. It's actually a really great phone, just a shame they can't release the source code sooner than this.

mamilux, 2018-08-29:
Already there!!

NhatHuyNguyen, 2018-08-29:
Problem solved :D

1nfrag, 2018-08-29:
Sweet. thanks Xiaomi for releasing the sources as requested!

closed 2018-08-29 by 1nfrag