MiCode/Xiaomi_Kernel_OpenSource #930: Redmi S2 missing touchscreen firmware

MacTavishAO, 2018-08-17:
Xiaomi, you are releasing kernel rarely at least do it RIGHT!

`fatal error: include/firmware/LQ_ysl_FT5446_Ofilm_EBBG_V0d_D01_20180403_app.i: No such file or directory
`

muralivijay, 2018-08-17:
lol xiaomi devs . idk why they always missing device important drivers . before redmi 5A battery driver issue . but now redmi s2 aka ysl focaltech firmware  driver missing . :D :D

tonystark009, 2018-08-17:
yes why are you cheating xiaomi, after became no.1 company head weight came uh??? What the hell!!! Dont chest karma is there

MacTavishAO, 2018-08-23:
Still don't give a shit ?

luk1337, 2018-08-23:
correct.

sushant9096, 2018-11-22:
drivers/input/touchscreen/focaltech_touch/focaltech_flash.c:50:28: fatal error: include/firmware/LQ_ysl_FT5446_Ofilm_EBBG_V0d_D01_20180403_app.i: No such file or directory
 #include FTS_UPGRADE_FW_APP
Xiaomi why you are dropped this firmware from source

dazza5000, 2020-01-07:
Did you find the firmware?

dazza5000, 2020-01-09:
I found the firmware in a kernel dump from a lenovo tb8504f