MiCode/Xiaomi_Kernel_OpenSource #935: gpl software license violation

TEF75, 2018-08-25:
Hello!

I have purchased phone Xiaomi Redmi 4 (prada)which is produced by your company and its firmware contains Linux, busybox etc, that is free software distributed under GNU General Public License. I would like to get the source code of the modified components of these programs, since the GPL requires you to release it. How I can do it?

skake, 2018-08-28:
Before opening a pool I suggest you look at the forums and the firmware for your phone is already available as well as the lineage 13

TEF75, 2018-08-29:
Я писал об открытом исходном коде и об его отсутствии, а не о кастовом роме

TEF75, 2018-08-29:
I wrote about the open source code and about its absence and not about the castom rom

PythonLimited, 2018-08-29:
Which for!Ware in particular? Seems that u don't know that MOST of the firmware is proprietary...

krasCGQ, 2018-09-18:
Look at santoni branch.