MiCode/Xiaomi_Kernel_OpenSource #939: Xiaomi Redmi Note 5A source code doesn't seem to build

mishaor, 2018-08-28:
```
In file included from drivers/net/ethernet/msm/rndis_ipa_trace.h:81:0,
                 from drivers/net/ethernet/msm/rndis_ipa.c:32:
include/trace/define_trace.h:83:43: fatal error: ./rndis_ipa_trace.h: No such file or directory
 #include TRACE_INCLUDE(TRACE_INCLUDE_FILE)
                                           ^
compilation terminated.
make[4]: *** [scripts/Makefile.build:257: drivers/net/ethernet/msm/rndis_ipa.o] Error 1
make[3]: *** [scripts/Makefile.build:402: drivers/net/ethernet/msm] Error 2
make[2]: *** [scripts/Makefile.build:402: drivers/net/ethernet] Error 2
make[1]: *** [scripts/Makefile.build:402: drivers/net] Error 2
make[1]: *** Waiting for unfinished jobs....
```

argraur, 2018-08-29:
Same issues here but with dipper source

mishaor, 2018-09-01:
Solved: you just need a patch, it's available here: https://gitlab.com/postmarketOS/pmbootstrap/blob/master/aports/device/linux-xiaomi-santoni/03_fix_rndis_ipa_trace_header_not_found.patch
Also there's another patch if it still doesn't compile (apply with the previous one): https://gitlab.com/postmarketOS/pmbootstrap/blob/master/aports/device/linux-xiaomi-santoni/02_fix_msm_dba.patch

closed 2018-09-01 by mishaor