MiCode/Xiaomi_Kernel_OpenSource #943: Redmi 5 Plus vince-o-oos

abdoi94, 2018-09-03:
Hi Vince/Redmi 5 Plus was updated to O on stable branch aka Miui 9.6

so please update kernel source accordingly

rama982, 2018-09-03:
yes, we need oreo kernel source

1chinke, 2018-09-03:
Without kernel source device won't be supported well by community. Please publish the kernel source code.

Blacksuan19, 2018-09-03:
WE NEED SAUCE BITCHES!!!

EnesSastim, 2018-09-03:
up xd

motoralbi, 2018-09-03:
U0

4PERTURE, 2018-09-03:
My email has been successfully spammed

EriecTan, 2018-09-03:
UP UP UP

dipanjandas1234, 2018-09-03:
Yes we need oreo kernel source

EriecTan, 2018-09-05:
what you waiting for m8?

abdoi94, 2018-09-05:
Vince O kernel source :\

1chinke, 2018-09-05:
C'mon

doctord98618, 2018-09-05:
what r u waiting for, give us vince sources already, development  stuck, don't wait nomore

EriecTan, 2018-09-08:
gPL sOfTwArE LiCeNsE ?

4PERTURE, 2018-09-08:
We're waiting

EriecTan, 2018-09-19:
gib sauce, git luv

EriecTan, 2018-09-24:
now we got miui 10 stable, so please release source code asap. thanks

laststandrighthere, 2018-09-25:
waiting for it

doctord98618, 2018-09-26:
So how much time you guys gonna take? Release sources..

rama982, 2018-09-26:
Its time for releasing source code.

EriecTan, 2018-09-26:
HeHao, release please

EriecTan, 2018-10-01:
still waiting..

doctord98618, 2018-10-01:
Thats sad, why this delay?, don't ignore issue pls

frocks, 2018-10-14:
🕒 JUST DO IT . 🕒

EriecTan, 2018-10-18:
*really* **dude**?

Bx16, 2018-10-22:
please update sources

Z5X67280, 2018-10-23:
Some intersting thing, MIUI Global version (based Android O) is use "daisy" kernel src.😂
so...... **We should be Wait to MIUI China version update to Android O.Although we have to accept this reality.** 😂

0312birdzhang, 2018-11-07:
It's time to release **vince-o-oos** now, as XiaoMi released the stable MIUI China O version .

EriecTan, 2018-11-07:
> It's time to release **vince-o-oos** now, as XiaoMi released the stable MIUI China O version .

really? i think its not released yet

rama982, 2018-11-07:
Yeah it still closed beta for Miui China oreo

rama982, 2018-11-09:
Today they released public beta for miui china oreo, hope they release **vince-o-oss** soon

EriecTan, 2018-11-09:
omfg im so hyped

laststandrighthere, 2018-11-26:
asdasdasd waiting for

rama982, 2018-11-26:
maybe we must waiting for tiffany o oss, vince kernel unified in there

EriecTan, 2018-12-05:
I hope its gonna be christmas gift

rama982, 2018-12-14:
its mid desember and still no **vince-o-oss** 
@liuguo09

EriecTan, 2018-12-14:
Please dont ignore

Z5X67280, 2018-12-14:
open src is long long way...

scafroglia93, 2018-12-14:
We hate communism but at the same time we adore the Chinese business model because prices are low

EriecTan, 2018-12-16:
give source please

0312birdzhang, 2018-12-21:
> It's time to release **vince-o-oos** now, as XiaoMi released the stable MIUI China O version .

This time it is true. 😂

EriecTan, 2018-12-21:
Yes, please

rama982, 2018-12-25:
Xiaomi already release miui china oreo stable to common user.
So its time to they release kernel source.

EriecTan, 2018-12-25:
Tag the dev please

Z5X67280, 2018-12-25:
What are you doing, xiaomi?
Where is GPLv2 license?
Are you eating GPLv2?
I suspect that Xiaomi's computers is still running Ubuntu with Pentium 2 (laugh)
Even Intel 386... (laugh)

_Send from [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/) for Redmi 5 Plus on Android P_

addhaloka, 2018-12-26:
> What are you doing, xiaomi?
> Where is GPLv2 license?
> Are you eating GPLv2?
> I suspect that Xiaomi's computers is still running Ubuntu with Pentium 2 (laugh)
> Even Intel 386... (laugh)
> 
> _Send from [FastHub-Libre](https://github.com/thermatk/FastHub-Libre/) for Redmi 5 Plus on Android P_

Xiaomi is working on Android Q

scafroglia93, 2019-01-10:
You complain and continue to support this shit producer. you are very strange

Z5X67280, 2019-01-21:
Well done, guys. [sakura-o-oss](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/sakura-o-oss).

abdoi94, 2019-01-22:
Closed sources were pushed

closed 2019-01-22 by abdoi94