MiCode/Xiaomi_Kernel_OpenSource #945: Problem with Audio and VIdeo Recording on POCO F1

AlekDevOfficial, 2018-09-08:
Compiled clean source for POCO F1 but Audio and Video Recording are broken please fix them

Thanks

rigaz29, 2018-09-09:
log ?

AlekDevOfficial, 2018-09-09:
Here's Dmesg log

AlekDevOfficial, 2018-09-09:
https://ufile.io/9lj85

AlekDevOfficial, 2018-10-01:
They released now  audio driver 2 DAYS ago but got compile errors so i fixed the broken audio but video recording bug is still here....

0xXA, 2018-10-10:
> They released now audio driver 2 DAYS ago but got compile errors so i fixed the broken audio but video recording bug is still here....

It was easy to fix I don't see any problems when I have fixed it

scafroglia93, 2018-10-10:
Are you the most xiaomi fanboy in the world ?

Please explain me why Xiaomi doesn't mantain without any rebase, is this the real communityz ?

Milestone everywere

0xXA, 2018-10-10:
> Are you the most xiaomi fanboy in the world ?
> 
> Please explain me why Xiaomi doesn't mantain without any rebase, is this the real communityz ?
> 
> Milestone everywere

First of all I'm not a fanboy mate!
Well there's not one device that Xiaomi is working on currently ( no one knows ) and another reason for this is they're retarded to do so, whenever a new kernel patchset releases comes with alot of commits and some of them aren't meant to be good for your device ( your soc ) or breaks something in your current kernel, and rebasing takes a lot of time if caf tag you're rebasing on have commits that changed the code that they have modified earlier in this situation it can take a lot of time to just import changes ( depends upon what's in your kernel currently )

scafroglia93, 2018-10-10:
Only apologies...