MiCode/Xiaomi_Kernel_OpenSource #949: Would it be possible to rebase Santoni's kernel? / 是否有可能改变 Santoni 的内核?

jrmoore, 2018-09-26:
Hi, Santoni's kernel is currently based on the LA.UM.5.6.r1-**04600**-89xx.0 tag, from May of 2017.

While it is still a 3.18 I wonder if it could be rebased to the LA.UM.5.6.r1-**08900**-89xx.0 tag from this August since it still targets the current Android version we're running, 7.1.2.

There's a newer 3.18 kernel that could be possibly more interesting, 3.18.72 in the LA.UM.**6.6.r1-09500**-89xx.0 tag and while that one is meant for 8.1 it should work just fine with 7.1.2.

Moving to a newer LTS kernel would be preferable, 4.4 (Nougat supported), 4.9, 4.14 but I know that's likely out of the question since it would require an Android upgrade to Oreo.

Thank you anyway for your time and your efforts, and if you needed help with it drop me a line.

Kind regards.

---

嗨, Santoni 的内核目前基于 LA.UM.5.6.r1-**04600**-89xx.0 标签, 从2017年5月开始.

虽然它仍然是 3.18, 我想知道它是否可以从今年8月开始重新定位到 LA.UM.5.6.r1-**08900**-89xx.0 标签, 因为它仍然针对我们正在运行的当前 Android 版本, 7.1.2.

还有一个更新的3.18内核可能更有趣 LA.UM.**6.6.r1-09500**-89xx.0  标签, 而那个是8.1的意思, 它应该可以正常使用 7.1.2.

转向更新的 LTS 内核将更可取, 4.4 (支持 Nougat), 4.9, 4.14, 但我知道这可能是不可能的, 因为它需要 Android 升级到 Oreo.

无论如何, 谢谢你的时间和努力, 如果你需要帮助, 请给我留言.

亲切的问候.

Haseo97, 2018-10-01:
dreams are desires (2)

0xXA, 2018-10-10:
> Hi, Santoni's kernel is currently based on the LA.UM.5.6.r1-**04600**-89xx.0 tag, from May of 2017.
>
> While it is still a 3.18 I wonder if it could be rebased to the LA.UM.5.6.r1-**08900**-89xx.0 tag [...]

Yes!, you can yourself rebase kernel to a newer tag for 3.18 and later upstream it!

Well, unfortunately it's not easy as your say to boot a 4.x kernel on 3.x as it needs a patched bootloader and comparable device drivers and patched blobs, but yes definitely you can rebase onto caf's new tags by importing your device specific changes on to it!

Thanks!
Yuvraj Saxena