MiCode/Xiaomi_Kernel_OpenSource #950: Release the Source code of Redmi Note 2 and of other MTK devices

Azizb750, 2018-09-30:
You have promised to release the kernel source of any device but I see that you have released only the kernel source Qualcomm devices only.
To be honest, we have fed up to waiting your Xiaomi to release the kernel source of MTK devices.
About Redmi Note 2 (and not only), there have been more than 3 years and still we are waiting to release the kernel source.

Please release the source code of our MTK devices, otherwise I will find a way to lawsuit your Company alongside Longcheer and Mediatek and I'm very very serious for it.

Thank you!

DeadneM, 2018-09-30:
Are you serious? You don't even know why mtk devices doesn't have kernel sources released... And you want to lawsuit a company for this? 
They don't have the source from that time mtk were not open source compliant at all 
So calm down or i will ban your from github I'm very serious for it.

Azizb750, 2018-09-30:
But are you sure that Xiaomi Does not have access to the kernel? If does not have access to it, how do they compile the kernel, check the miui kernel version if you want

Azizb750, 2018-09-30:
I see that the person who compile it its xiaomi cause I see that both host and user seems to belong to xiaomi's pc

"Linux version 3.10.61+8c25ca99-3ffb4-8564dbd9 (builder@c4-miui-ota-bd15.bj) (gcc version 4.9 20140514 (mtk-20150408) (GCC)) #1 SMP PREEMPT Wed Aug 1 20:22:31 CST 2018" 

And you still believe that they dont have access to the source code?? @DeadneM

PythonLimited, 2018-09-30:
Well that doesnt mean that they are using a source code compiled kernel.
This variable is generated automatically by the build system.
Could be that xiaomi is indeed using a pre built kernel from MTK.

Dinolek, 2018-09-30:
@PythonLimited They have full access to mtk alps.
/dev/xiaomi_memsw_state is there just like in other xiaomi devices.
https://github.com/jianglu/framework_native
Xiaomi developer posted source code of frameworks/native that include both odm and MIUI changes.

scafroglia93, 2018-10-10:
Why BQ provide full mtk kernel source ?

Please, ban me from github LOL

0xXA, 2018-10-10:
> Why BQ provide full mtk kernel source ?
> 
> Please, ban me from github LOL

They wanted to release and they did any problems?
Xiaomi is a hoe

luk1337, 2018-10-10:
i wish they'd do so.

Azizb750, 2018-10-10:
> 
> 
> > Are you serious? You don't even know why mtk devices doesn't have kernel sources released... And you want to lawsuit a company for this?
> > They don't have the source from that time mtk were not open source compliant at all
> > So calm down or i will ban your from github I'm very serious for it.
> 
> Well not mtk but your vendor can release sources if they pay for this to mtk!
> So, don't be a smartass
> 
> > Well that doesnt mean that they are using a source code compiled kernel.
> > This variable is generated automatically by the build system.
> > Could be that xiaomi is indeed using a pre built kernel from MTK.
> 
> For Your Kind Information whenever a vendor makes a deal with soc manufacturers( mtk, qcom, etc) for their soc than soc manufactures also provides a manual on how to compile it + sources for soc they're dealing with! No matter if it's mtk or qcom, vendor will definitely get sources for their dealed socs!
> And (mtk, qcom, etc) will not compile kernels for you, vendor have to do that, mtk provides only Alps sources which contains sources for many libraries as well as kernel also ( you can say it as a "Complete Package" ) job of your vendor is to just add your device specific changes on it!
> And also the date on which kernel is compiled is not changeable or if you understand binary code than yes it's, mostly nobody have this much knowledge!
> 
> > I see that the person who compile it its xiaomi cause I see that both host and user seems to belong to xiaomi's pc
> > "Linux version 3.10.61+8c25ca99-3ffb4-8564dbd9 ([builder@c4-miui-ota-bd15.bj](mailto:builder@c4-miui-ota-bd15.bj)) (gcc version 4.9 20140514 (mtk-20150408) (GCC)) #1 SMP PREEMPT Wed Aug 1 20:22:31 CST 2018"
> > Edit: this is from miui 9.6.3.0 prebuilt kernel
> > And you still believe that they dont have access to the source code?? @DeadneM
> 
> Here is a list of drivers I got when I reversed some code from your stock firmwares
> https://del.dog/raw/bexibatici
how did you reversed the drivers? is there any good way to reverse prebuilt kernel?
i have ida pro installed but I cant dissasemble the kernel for a reason, somewhere i have a serious mistake
> > Here's a Kernel for Hermes that's made by reversing code by @nofearnohappy
> > https://github.com/Azizb750/android_kernel_xiaomi_hermes
 about this, I had forked his kernels so i can improve them in the future if possible, cause I'm interested to reverse engineering a lot, yeah when I asked him on telegram about how reverse engineering helped him, he said that he helped him a lot but he needs a script 
need an intelligent person who creates a script for IDA which can convert to the inverse code, so, can you tell us how you did this please? did you reversed by hand or did you used an extention on ida?

0xXA, 2018-10-10:
> how did you reversed the drivers? is there any good way to reverse prebuilt kernel?
> i have ida pro installed but I cant dissasemble the kernel for a reason, somewhere i have a serious mistake [...]

Well these paths are saved as data inside binary, you can later analyse the whole binary for it, and yes you can't disassemble a 64bit kernel easily there is a lot of problems in doing so. Actually, ida gives wrong code for arm64 by default and it's fixed in ida 7.1 , but you can still disassemble it , by your message I think you're talking about loading symbols from kallsyms file to binary than yes you can but you need to remove  headers containing dtb sections, I did this manually by hands and literally no script will convert and give you exact code to put in driver, you have to do it by hands by analysing the function which is required sriots are just to parse external symbols into your binary

Azizb750, 2018-10-11:
Yeah, I know that @nofearnohappy didn't reversed it, but he says that it helped a lot to look the functions of the compiled drivers in the kernel

heindrix, 2018-12-26:
> Are you serious? You don't even know why mtk devices doesn't have kernel sources released... And you want to lawsuit a company for this?
> They don't have the source from that time mtk were not open source compliant at all
> So calm down or i will ban your from github I'm very serious for it.

LOLOL.
You can try ban me, i'm waiting for that to be happened, i'll wait for a week, i think that's enough time to do that.

You should write your opinion with more composure, rather than boasting like that.
We are all in here is to give contribution to the community, not to boasting each other.

Good day and merry christmas. Ho Ho Ho

dragonpt, 2018-12-28:
C'mon!!?

People still blaming MediaTek?

Even back 4/5 years ago...  The responsibility to release kernel source relies on the OEM or ODM ...
Truth is many ODMs don't even give kernel source to their clients, even when requested 

But what do I know right?

santomet, 2019-01-16:
Thank Youuuu, Mi 😍 not that hard, was it?

Azizb750, 2019-01-17:
I will close now the issue as hermes and hennessy's kernel source is now released, thanks @Micode a lot, please keep releasing the kernel source of other MTK devices too, have a good year

scafroglia93, 2019-01-17:
3 years to relase a lollipop source 

LOL

#chinesefuckgpl

dragonpt, 2019-01-17:
Still better then the rest of ~90% MediaTek devices by small/medium ODMs...

ryan-andri, 2019-01-17:
> Still better then the rest of ~90% MediaTek devices by small/medium ODMs...

diz.

scafroglia93, 2019-01-17:
I do not care what others do; is it ever possible that there is always a fucking excuse for this manufacturer?

The OEM that supports me does not have these problems to release a maintained kernel source and not a fucking snapshot loaded and left to rot

dragonpt, 2019-01-17:
Of course, there's no excuses for not releasing it, or delaying a released to the point of being useless ( 3 years later)

GPL is something many manufacturers ( mainly) chineses ones, don't care about

Cheers

dragonpt, 2019-01-17:
Now, with the growth of Xiaomi globally, the newer partnerships, and with the help of some new employees ( that do care) and some people on " the management " , things are now becoming slightly better on Xiaomi GPL, yet, I know very well it's still not perfect... But you do recon things are way better then like ~1 year ago?

nofearnohappy, 2019-01-17:
4 years later

Azizb750, 2019-02-10:
Hope That @Micode would release kernel source of other MTK Devices too (like Redmi Note 4 MTK)

closed 2019-02-10 by Azizb750