MiCode/Xiaomi_Kernel_OpenSource #957: 求大佬发编译教程

chn-lee-yumi, 2018-10-20:
我跟着wiki做了……（花了一天时间下了100多G的开发环境），编译的时候报错，google了一下发现是sdclang的问题，修复之后继续编译，编译的时候报了几个undefined error，我注释掉后编译成功，刷机后启动报错。
求个没坑的教程……

stormeyes, 2018-11-02:
如果你只是想编译内核并刷入, 不用按照wiki步骤去下载几百G的东西, 只需要下载小米开源的这部分内核源码就已经足够让你编译一个可以自定义并刷入自己的ROM的内核. 按照下面的步骤来执行. 这里以我的手机 mi max3, 编译环境 ubuntu 16.04 为例:

+ 安装开发环境相关包
```
sudo apt install -y bc bison build-essential ccache curl flex g++-multilib gcc-multilib git gnupg gperf imagemagick lib32ncurses5-dev lib32readline-dev lib32z1-dev liblz4-tool libncurses5-dev libsdl1.2-dev libssl-dev libwxgtk3.0-dev libxml2 libxml2-utils lzop pngcrush rsync schedtool squashfs-tools xsltproc zip zlib1g-dev unzip openjdk-8-jdk language-pack-zh-hans
```
+ checkout 内核源码
```
git clone https://github.com/MiCode/Xiaomi_Kernel_OpenSource -b nitrogen-o-oss
```

+ 下载跨平台编译链工具
```
git clone https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9
```

+ 编译前的环境变量配置
```
export ARCH=arm64
export SUBARCH=arm64
export CROSS_COMPILE=/home/stormeye/aarch64-linux-android-4.9/aarch64-linux-android-
```
这里有两点要注意的:
CROSS_COMPILE 这里的路径请记得修改为你自己的路径
CROSS_COMPILE 的路径最后有一个 **-** , 至于为什么, 看一下这个路径下的二进制文件的名字规则, 和Makefile是怎么写的就明白了

+ 查找并确定 defconf 文件, 这个文件的路径一般在内核源码的 `arch/arm64/configs` 目录下, 会有一大堆的 defconf 文件, 找到最适配你的机型的 defconf 文件, 我这里是 nitrogen_user_defconfig

+ 进入内核源码的根目录并开始编译
```
make O=out nitrogen_user_defconfig
make O=out -j8
```
 这里要注意, O=out 不能省略, 否则会出现类似 `include` 尖角括号找不到头文件 这种问题 #https://github.com/OnePlusOSS/android_kernel_oneplus_msm8998/issues/3 一般来说小米代码开源前都已经走过流程, 是可以正常编译成功的

+ 编译 success 后, 会在 `out/arch/arm64/boot` 目录下生成 `Image.gz-dtb` 文件, 这就是编译出来的内核文件

+ 使用AnyKernel2 #https://github.com/osm0sis/AnyKernel2 , 按照说明替换你的`Image.gz-dtb` 文件并编辑 `anykernel.sh` 文件. 里面的参数非常非常非常简单, 按照自己的实际需求编辑一下里面的参数并打包成zip文件. 或者更简单的, 你找一个你的设备的第三方内核zip包, 建议在不解压的情况下直接删除zip包里面的 `Image.gz-dtb` 并替换成你的 `Image.gz-dtb` 文件即可.

+ 进入twrp, 刷入这个zip文件, 开机. 如果出问题, `anykernel.sh`里面可以配置日志的路径, 你可以查看日志相关的信息.

+ 如果你想将其打包成 `boot.img` 文件, 上小米官网下载手机对应的官方线刷包, 解压出来在images目录下有 boot.img, 使用 unpackbootimg 工具将这个img解开, 把你的`Image.gz-dtb`替换进去并用 mkbootimg 重新打包成 boot.img 即可.

+ 注意, 刷入后你的 WIFI 可能不能正常工作, 原因和解决方案在这里: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/FAQ
 
下面是我新鲜编译的内核:
[image]

changxiangzhong, 2018-11-29:
加精吧，写的真好。

1361189276, 2018-11-30:
虽然看不懂 但是很喜欢看

Gah0, 2018-12-02:
[https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/979](url)
有谁能知道这个怎么解决，Warning

DroidKali, 2018-12-05:
小米官方开源的源码里面代码都是错的，不会C不修改源码改正里面的错误代码，能编译成功都是做梦，不要抱有太大的希望，小米的开源只是伪开源罢了

ivan19871002, 2018-12-06:
> 小米官方开源的源码里面代码都是错的，不会C不修改源码改正里面的错误代码，能编译成功都是做梦，不要抱有太大的希望，小米的开源只是伪开源罢了

我们开源的代码都是基于完整的android编译环境下编译的，我们本地编译并且验证过再开源的，不懂就不要瞎说。 你单独编译kernel 还有用的gcc 版本也不一样肯定是解决环境问题

DroidKali, 2018-12-06:
@ivan19871002 呵呵，我不懂？要是你们的源代码纯净还需要别人来给你们打patch？还有kernel不是本来就要单独编译的？你真的了解过linux kernel？GCC版本我不明白难道我用谷歌官方4.9的GCC版本也不对？按照谷歌官方的环境配置也不行？谷歌官方的环境是错的了？我一直以来学的是个假内核编译？

ivan19871002, 2018-12-10:
> 我跟着wiki做了……（花了一天时间下了100多G的开发环境），编译的时候报错，google了一下发现是sdclang的问题，修复之后继续编译，编译的时候报了几个undefined error，我注释掉后编译成功，刷机后启动报错。
> 求个没坑的教程……

二楼的教程就行了，内部开发kernel都是跟着系统在一起的，所以在kernel编译配置上都是有现成的配置的。 
如果单独编译kernel，我是按照下面配置来做的，例如我想编译MAX 3的内核

主机 ：  ubuntu 18.04
手机： Xiaomi Max 3
gcc 工具，Android P源码内自带

export ARCH=arm64
export SUBARCH=arm64
export CROSS_COMPILE=/code/android-p/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-
mkdir out
make O=out nitrogen_user_defconfig
make -j24 O=out
机器性能好的话，-j24可以更多 比如-j32  -j64

编译结束之后，会在 source code 目录下面的out目录生成编译好的symbol 文件
out/arch/arm64/boot
这个目录下面会有你需要的kernel 编译文件。
另外如果烧上去wlan 不能使用，或者 其他平台audio不能使用的话（例如sdm845增加了techpack/audio 目录 ，这个在源码里面单独是一个project，开源的时候跟内核一起开源了），也需要把对应的 源码编译出module push到手机里面。

大家经常编译出问题的找不到头文件，或者<>要改成""  是因为高通在书写代码的时候，默认有引用路径，有O 这个选项，这个选项在 kernel 的makefile里面有说明。如果定义了O在编译的时候引用头文件就会准确，没有定义的话，默认可能就会失败。

closed 2018-12-10 by ivan19871002

chn-lee-yumi, 2019-06-01:
我还踩了个坑，就是配置没选对……换个配置就编译成功了

chn-lee-yumi, 2019-06-01:
另外，编译出Image.gz-dtb之后，我是用abootimg来替换内核。首先用TWRP备份boot.img，然后再用abootimg替换：
```
abootimg -u boot.img -k /root/Image.gz-dtb 
```
最后复制到手机上，用TWRP直接刷入boot.img

wode2016501, 2019-10-01:
小米6 怎么选defconf

wode2016501, 2019-10-01:
小米6 怎么选defconf

wangcong15, 2020-03-07:
想请教一下，该怎么对开发版ROM的手机上刷内核，我同时还想保留ROOT...

cok3307788, 2020-05-10:
> 想请教一下，该怎么对开发版ROM的手机上刷内核，我同时还想保留ROOT...

要刷入内核以后再刷入magisk.zip
PS:提个坑，dtb和dtbo.img不一致会开不了机，可以用二进制工具硬提取旧内核的dtb

shandongtlb, 2020-05-10:
> #979
> 有大佬知道这个怎么解决，Warning。

哇发现了yellow gay大佬

Snow-tear, 2020-10-24:
你好, 请问wifi驱动怎么使用?

Ray1108, 2020-12-09:
> > 我跟着wiki做了……（花了一天时间下了100多G的开发环境），编译的时候报错，google了一下发现是sdclang的问题，修复之后继续编译，编译的时候报了几个undefined error，我注释掉后编译成功，刷机后启动报错。
> > 求个没坑的教程……
> 
> 二楼的教程就行了，内部开发kernel都是跟着系统在一起的，所以在kernel编译配置上都是有现成的配置的。
> 如果单独编译kernel，我是按照下面配置来做的，例如我想编译MAX 3的内核
> 
> 主机 ： ubuntu 18.04
> 手机： Xiaomi Max 3
> gcc 工具，Android P源码内自带
> 
> export ARCH=arm64
> export SUBARCH=arm64
> export CROSS_COMPILE=/code/android-p/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9/bin/aarch64-linux-android-
> mkdir out
> make O=out nitrogen_user_defconfig
> make -j24 O=out
> 机器性能好的话，-j24可以更多 比如-j32 -j64
> 
> 编译结束之后，会在 source code 目录下面的out目录生成编译好的symbol 文件
> out/arch/arm64/boot
> 这个目录下面会有你需要的kernel 编译文件。
> 这里面的binary不能直接烧到手机上，需要打包成boot.img， 另从某些版本开始也需要用到dtbo，所以也需要打包对应的dtbo.img。 具体如何把编译出来的kernel binary打包成boot.img 和 dtbo.img请Google找一下答案，xda论坛有很多种方法可以使用。
> 
> 另外如果烧上去wlan 不能使用，或者 其他平台audio不能使用的话（例如sdm845增加了techpack/audio 目录 ，这个在源码里面单独是一个project，开源的时候跟内核一起开源了），也需要把对应的 源码编译出module push到手机里面。
> 
> 大家经常编译出问题的找不到头文件，或者<>要改成"" 是因为高通在书写代码的时候，默认有引用路径，有O 这个选项，这个选项在 kernel 的makefile里面有说明。如果定义了O在编译的时候引用头文件就会准确，没有定义的话，默认可能就会失败。

大佬，为啥我编译的只生成image.gz而不生成image.gz-dtb呢？

Ray1108, 2020-12-09:
> 如果你只是想编译内核并刷入, 不用按照wiki步骤去下载几百G的东西, 只需要下载小米开源的这部分内核源码就已经足够让你编译一个可以自定义并刷入自己的ROM的内核. 按照下面的步骤来执行. 这里以我的手机 mi max3, 编译环境 ubuntu 16.04 为例:
> 
> * 安装开发环境相关包
> 
> ```
> sudo apt install -y bc bison build-essential ccache curl flex g++-multilib gcc-multilib git gnupg gperf imagemagick lib32ncurses5-dev lib32readline-dev lib32z1-dev liblz4-tool libncurses5-dev libsdl1.2-dev libssl-dev libwxgtk3.0-dev libxml2 libxml2-utils lzop pngcrush rsync schedtool squashfs-tools xsltproc zip zlib1g-dev unzip openjdk-8-jdk language-pack-zh-hans
> ```
> 
> * checkout 内核源码
> 
> ```
> git clone --depth=1 https://github.com/MiCode/Xiaomi_Kernel_OpenSource -b nitrogen-o-oss
> ```
> 
> * 下载跨平台编译链工具
> 
> ```
> git clone --depth=1 https://android.googlesource.com/platform/prebuilts/gcc/linux-x86/aarch64/aarch64-linux-android-4.9
> ```
> 
> * 编译前的环境变量配置
> 
> ```
> export ARCH=arm64
> export SUBARCH=arm64
> export CROSS_COMPILE=/home/stormeye/aarch64-linux-android-4.9/bin/aarch64-linux-android-
> ```
> 
> 这里有两点要注意的:
> CROSS_COMPILE 这里的路径请记得修改为你自己的路径
> CROSS_COMPILE 的路径最后有一个 **-** , 至于为什么, 看一下这个路径下的二进制文件的名字规则, 和Makefile是怎么写的就明白了
> 
> * 查找并确定 defconf 文件, 这个文件的路径一般在内核源码的 `arch/arm64/configs` 目录下, 会有一大堆的 defconf 文件, 找到最适配你的机型的 defconf 文件, 我这里是 nitrogen_user_defconfig
> * 进入内核源码的根目录并开始编译
> 
> ```
> make O=out nitrogen_user_defconfig
> make O=out -j8
> ```
> 
> 这里要注意, O=out 不能省略, 否则会出现类似 `include` 尖角括号找不到头文件 这种问题 #[OnePlusOSS/android_kernel_oneplus_msm8998#3](https://github.com/OnePlusOSS/android_kernel_oneplus_msm8998/issues/3) 一般来说小米代码开源前都已经走过流程, 是可以正常编译成功的
> 
> * 编译 success 后, 会在 `out/arch/arm64/boot` 目录下生成 `Image.gz-dtb` 文件, 这就是编译出来的内核文件
> * 使用AnyKernel2 #https://github.com/osm0sis/AnyKernel2 , 按照说明替换你的`Image.gz-dtb` 文件并编辑 `anykernel.sh` 文件. 里面的参数非常非常非常简单, 按照自己的实际需求编辑一下里面的参数并打包成zip文件. 或者更简单的, 你找一个你的设备的第三方内核zip包, 建议在不解压的情况下直接删除zip包里面的 `Image.gz-dtb` 并替换成你的 `Image.gz-dtb` 文件即可.
> * 进入twrp, 刷入这个zip文件, 开机. 如果出问题, `anykernel.sh`里面可以配置日志的路径, 你可以查看日志相关的信息.
> * 如果你想将其打包成 `boot.img` 文件, 上小米官网下载手机对应的官方线刷包, 解压出来在images目录下有 boot.img, 使用 unpackbootimg 工具将这个img解开, 把你的`Image.gz-dtb`替换进去并用 mkbootimg 重新打包成 boot.img 即可.
> * 注意, 刷入后你的 WIFI 可能不能正常工作, 原因和解决方案在这里: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/FAQ
> 
> 下面是我新鲜编译的内核:
> [image]

大佬，为啥我编译的只生成image.gz而不生成image.gz-dtb呢？

jhksyghr, 2021-01-18:
> 大佬，为啥我编译的只生成image.gz而不生成image.gz-dtb呢？
请问你找到原因了吗？ 我也遇到了只生成image.gz而不生成image.gz-dtb:(

han1985, 2021-02-15:
小米内核编译 一坨屎，照着Wiki 能编译好 见鬼呢。真TM 烂。

Cussrro, 2021-05-14:
我也觉得，错了一晚上，照着他们说的做编译不了，小米内核编译 一坨屎，照着Wiki 能编译好 见鬼呢。真TM 烂。

sesame-foxes, 2022-02-05:
编译倒是完成了，wifi太恶心了吧，已经开机一天下了一百多g了，还没跑完。

Hipye, 2022-02-19:
**内核没问题**，大多数人的问题就是gcc版本的问题。这里提个小建议### 去看内核最早的commit时间，然后去googlesource clone比提交时间早几个月的版本的gcc。😂😂

ruje0504, 2022-04-17:
联发科的u没有驱动咋搞

gdannyliao, 2022-04-27:
> 

这明显是经验之谈

gdannyliao, 2022-04-27:
> **内核没问题**，大多数人的问题就是gcc版本的问题。这里提个小建议### 去看内核最早的commit时间，然后去googlesource clone比提交时间早几个月的版本的gcc。😂😂

这明显是经验之谈

TinyMagicka, 2023-06-12:
很多人clone了交叉编译的gcc之后发现里面工具都被删了或编译各种报错, 可以将gcc项目换到之前的版本 
git reset --hard 7f0c44b

Alex6357, 2023-09-20:
> 另外如果烧上去wlan 不能使用，或者 其他平台audio不能使用的话（例如sdm845增加了techpack/audio 目录 ，这个在源码里面单独是一个project，开源的时候跟内核一起开源了），也需要把对应的 源码编译出module push到手机里面。

这步要怎么操作？

lzyfirst, 2023-10-11:
哈哈哈，23年的回眸

jianyi520, 2024-03-25:
> **内核没问题**，大多数人的问题就是gcc版本的问题。这里提个小建议### 去看内核最早的commit时间，然后去googlesource clone比提交时间早几个月的版本的gcc。😂😂
找不到在哪clone 求教

UncleLiuYe, 2024-05-04:
> 大佬，为啥我编译的只生成image.gz而不生成image.gz-dtb呢？

> 如果你只是想编译内核并刷入, 不用按照wiki步骤去下载几百G的东西, 只需要下载小米开源的这部分内核源码就已经足够让你编译一个可以自定义并刷入自己的ROM的内核. 按照下面的步骤来执行. 这里以我的手机 mi max3, 编译环境 ubuntu 16.04 为例:
>
> * 安装开发环境相关包
>
> ```
> sudo apt install -y bc bison build-essential [...]

如果你编译的内核版本大于等于4.19，那么他只会编译出一个Image，dtbo什么的需要手动打包，请了解mkdtboimg脚本