MiCode/Xiaomi_Kernel_OpenSource #962: Whyred huaxing panel

akhilnarang, 2018-11-03:
The current sources don't work on devices with huaxing nt36672 panel, we just get a black screen
Kindly update the source for it

faiyazsheth, 2018-11-03:
@hehao

DevanshBajaj, 2018-11-03:
^^^^

SentinelX, 2018-11-03:
Update source!!!

Meraj018, 2018-11-03:
@hehao

Meraj018, 2018-11-03:
Pls xiaomi update whyred source..

IntelliJAbhishek, 2018-11-03:
Xiaomi Please update the sources. Support us for custom rom development

GuaiYiHu, 2018-11-05:
Having informed the developer.

faiyazsheth, 2018-11-09:
@hehao :-|

NiNoXowl, 2018-11-15:
up

dileepxdn, 2018-11-19:
up

Tao-kun, 2018-11-23:
Is [this](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/whyred-o-oss/arch/arm/boot/dts/qcom/dsi-panel-huaxing-nt36672-1080p-video.dtsi) the driver of huaxing panel?

NiNoXowl, 2018-12-01:
up up

Haseo97, 2018-12-01:
@hehao bdsk hear us here ;-;

0xXA, 2018-12-01:
Is this issue fixed in recent xiaomi firmwares?

Haseo97, 2018-12-02:
This commit fixed the panel for huaxing: https://github.com/psndna88/AGNi_pureMIUI/commit/ce73a14bf988280fbd535968ef4819b74fbdf296

closed 2018-12-03 by akhilnarang

buburoul, 2018-12-04:
Nahi degi (2)