MiCode/Xiaomi_Kernel_OpenSource #975: Put kernel source of Redmi 6 and 6A

SamarV-121, 2018-11-19:


XC16, 2018-12-04:
agree,please release redmi 6's kernel source

rama982, 2018-12-04:
Its always hard to release mtk source

Dreyk007, 2018-12-10:
So, we still need Redmi 6A sources...

0xXA, 2018-12-10:
> Its always hard to release mtk source

No, as usual you can release your modified kernel but not mtk's ALPS sources. Like sources for Hals, lk, bootloader.
If they have released sources for one mtk device in any of the branches of this repo than you can appeal them to release for this device too

0xXA, 2018-12-10:
> Its always hard to release mtk source

No, as usual oem can release their modified kernel but not mtk's ALPS sources. Like sources for Hals, lk, bootloader,etc
If they want to release it too than have to pay extra for this.
If they have released sources for one mtk device in any of the branches of this repo than you can appeal them to release for this device too

closed 2018-12-25 by SamarV-121

Dreyk007, 2018-12-25:
Closed due to released Redmi 6/6A sources or what?

XC16, 2018-12-25:
ya it released

> Closed due to released Redmi 6/6A sources or what?