MiCode/Xiaomi_Kernel_OpenSource #978: missing include/firmware/LQ_ysl_FT5446_Ofilm_EBBG_V0d_D01_20180403_app.i

sushant9096, 2018-11-22:
drivers/input/touchscreen/focaltech_touch/focaltech_flash.c:50:28: fatal error: include/firmware/LQ_ysl_FT5446_Ofilm_EBBG_V0d_D01_20180403_app.i: No such file or directory
#include FTS_UPGRADE_FW_APP
Xiaomi why you are dropped this firmware from source

MacTavishAO, 2018-11-22:
Because they are idiots.

scafroglia93, 2018-11-27:
Sticazzi MIUI

0xXA, 2018-11-30:
> drivers/input/touchscreen/focaltech_touch/focaltech_flash.c:50:28: fatal error: include/firmware/LQ_ysl_FT5446_Ofilm_EBBG_V0d_D01_20180403_app.i: No such file or directory
> #include FTS_UPGRADE_FW_APP
> Xiaomi why you are dropped this firmware from source

Xiaomi is in another dimension, don't expect them to release that also. They don't even know what `.gitignore` does .i.e, in .gitignore file *.i literally matches all the files with *.i extension you can check it here

https://github.com/0xXA/Xiaomi_Kernel_OpenSource/blob/ysl-o-oss/.gitignore#L21

I reversed your firmware it's working fine, i hope you liked my work and also you can cherry-pick commit from my repo. Here,

https://github.com/0xXA/Xiaomi_Kernel_OpenSource/commits/ysl-o-oss

Thanks

0xXA, 2018-12-10:
#985
Fixes it