MiCode/Xiaomi_Kernel_OpenSource #979: wlan.ko bug in miui

Gah0, 2018-11-27:
how everything is going
```
11-27 20:16:41.358 D/WifiService( 2223): setWifiEnabled: true pid=27548, uid=1000
11-27 20:16:41.454 W/wlan    (    0): module is from the staging directory, the quality is unknown, you have been warned.
11-27 20:16:41.462 E/wlan    (    0): Unknown parameter `mac'
11-27 20:16:41.527 D/WifiHW  ( 2223): finit_module for "/system/lib/modules/wlan.ko" failed: No such file or directory
11-27 20:16:41.527 E/WifiStateMachine( 2223): Failed to load driver
```
dmesg.log
```
[10068.612677] wlan: module is from the staging directory, the quality is unknown, you have been warned.
[10068.632282] wlan: Unknown parameter `mac'
```

scafroglia93, 2018-11-27:
Ciao MAMMA SONO FAMOSO

Gah0, 2018-11-27:
phone=leo
qcacld-2.0
msm8994

stormeyes, 2018-12-02:
from the error message: 
```
finit_module for "/system/lib/modules/wlan.ko" failed: No such file or directory
```

Maybe u should try to put your wlan.ko to the path `/system/lib/modules/` and try again?

If the problem still occur, I recommend u directly compile it into kernel when u compile the kernel code instead of module.

Gah0, 2019-02-23:
catch .ko on miui10 try to use same wlan version driver
i strings qca_cld_wlan.ko on terminal
and wlan driver is `4.0.11.144`,

Gah0, 2019-02-28:
work now

closed 2019-02-28 by Gah0