MiCode/Xiaomi_Kernel_OpenSource #981: Redmi Note 6 pro (tulip) kernel sources

DrRamm, 2018-11-30:
Even mi 8 lite is now available

qttc, 2018-11-30:
please, publish sources!

sayaadali, 2018-11-30:
Yah please do it we really need it

sodacost, 2018-11-30:
+++ please

Rekembus, 2018-11-30:
+++

Niko8080, 2018-11-30:
Please publish the sources of the kernel.++++

ctopoj, 2018-11-30:
+++

ctopoj, 2018-11-30:
Please publish the sources of the kernel.++++

dadalife1, 2018-12-01:
Please publish the sources of the kernel.+++

hronus, 2018-12-03:
please publish sources! I like stock android!

lzzy12, 2018-12-08:
Please publish kernel sources respecting GPLv2

szltom, 2018-12-08:
Please publish kernel source

akwizard, 2018-12-08:
Please publish tulip kernel sources.++++

PrateekPunetha, 2018-12-08:
Bitches Release It :laughing:

taranhora, 2018-12-08:
We need kernel source to start the development of redmi note 6 pro (tulip)  so provide ASAP.

Daniel8821, 2018-12-09:
Please publish the sources of the kernel

ProfessorMi, 2018-12-10:
Please publish the tulip kernel sources

ux1, 2018-12-10:
Please publish kernel source.

msrshahkar, 2018-12-10:
Please publish the kernel sources. We love Xiaomi Devices but without kernel sources it's no fun.

thesprintster, 2018-12-10:
@ivan19871002 @hehao any ETA ?

AziKoo, 2018-12-10:
Please publish kernel sources++++

joiannariello, 2018-12-10:
please release kernel sources

mtpockets1, 2018-12-10:
Please publish the kernel source ASAP

rama982, 2018-12-10:
Please release tulip kernel source, and vince oreo kernel source too

PrateekPunetha, 2018-12-10:
@rama982😆😆😆 hahahah yesssss

athulharold, 2018-12-10:
Please publish the kernel Sources

davideriboli, 2018-12-10:
Please publish tulip kernel sources.

In this century.

Saksham033, 2018-12-12:
Man push the tulip source ffs. Look at other brands. They release the source within days of release... But you guys wanna delay it. Ffs release them...

kvdrrrrr, 2018-12-12:
no, it's not happening.

tankist8879, 2018-12-12:
Please publish tulip kernel sources.

DarkCoder0, 2018-12-16:
Please Publish Source Code..  Why you always Do that Xiaomi.. :/

thesprintster, 2018-12-19:
Please release kernel source.

thesprintster, 2018-12-20:
Its been **more than 3 months** since launch. Mi promised to publish source with 3 months.

**Please give us kernel source.**

thesprintster, 2018-12-20:
@Luk1337 are you buying this phone? ;)

Saw your down-vote on @Saksham033 comment.

thesprintster, 2018-12-22:
For the love of god give us tulip kernel source! :(

MrYacha, 2018-12-24:
Release sources. Respect GPLv2 licence

sayaadali, 2018-12-24:
Yes please release  it

sayaadali, 2018-12-24:
According to Xiaomi they said that they will release kernel source after 3 month of device lunche

daniil20, 2018-12-24:
Please publish the sources of the kernel:sob:

ihsanepixel, 2018-12-24:
Please publish the sources of the kernel.

ihsanepixel, 2018-12-24:
> According to Xiaomi they said that they will release kernel source after 3 month of device lunche

after the launch in India or the world?

Agneton, 2018-12-25:
Please publish the sources of the kernel

esashwin, 2018-12-25:
please publish sources......

rashedsahaji, 2018-12-25:
Please sir it's our request

rama982, 2018-12-25:
Even jasmine-p-oss have tulip driver on it lol.

Please xiaomi release full kernel source for tulip.

ryan-andri, 2018-12-25:
that's xiaomi @rama982

vitorindeep, 2019-01-03:
Would be lovely if you do publish the sources.

zignd, 2019-01-08:
Release the kernel source code please, I have never asked you anything.

rama982, 2019-01-08:
its released now

closed 2019-01-09 by DrRamm

satana666luciferka, 2019-01-29:
Кто может собрать ядро NETHUNTER please

SahilSonar, 2019-02-07:
PLEASE RELEASE IT!!! GUYS ITS FOR IT! NOW RN7 WILL BE OUT THEN WHY NOT JUST A SOURCE !!

rama982, 2019-02-07:
> PLEASE RELEASE IT!!! GUYS ITS FOR IT! NOW RN7 WILL BE OUT THEN WHY NOT JUST A SOURCE !!

Really m8? its already released inside whyred-o-oss (full changes, latest patch) and jasmine-p-oss (partial changes)