MiCode/Xiaomi_Kernel_OpenSource #983: compile error. No file xt_mark.h

artyom5613, 2018-12-01:
net/netfilter/xt_mark.c:16:38: fatal error: linux/netfilter/xt_hmark.h: No such file or directory
Error ^^^

 I go to file and find #include <linux/netfilter/xt_mark.h>

Go into /include/linux/netfilter/

and really don't find this file. Please fix.
(Daisy defconf, mi a2 lite kernel source)

artyom5613, 2018-12-01:
` make -j4
arch/arm64/Makefile:130: warning: overriding recipe for target 'vdso_prepare'
arch/arm64/Makefile:120: warning: ignoring old recipe for target 'vdso_prepare'
arch/arm64/Makefile:130: warning: overriding recipe for target 'vdso_prepare'
arch/arm64/Makefile:120: warning: ignoring old recipe for target 'vdso_prepare'
scripts/kconfig/conf  --silentoldconfig Kconfig
drivers/input/touchscreen/ft5446/Kconfig:5:warning: ignoring type redefinition of 'TOUCHSCREEN_FT5446' from 'tristate' to 'boolean'
drivers/input/touchscreen/gt917d/Kconfig:4:warning: ignoring type redefinition of 'TOUCHSCREEN_GT917d' from 'tristate' to 'boolean'
arch/arm64/Makefile:130: warning: overriding recipe for target 'vdso_prepare'
arch/arm64/Makefile:120: warning: ignoring old recipe for target 'vdso_prepare'
  CHK     include/config/kernel.release
  CHK     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  CALL    scripts/checksyscalls.sh
make[1]: 'include/generated/vdso-offsets.h' is up to date.
  CHK     include/generated/compile.h
  GZIP    kernel/config_data.gz
  CHK     kernel/config_data.h
  CC      drivers/media/platform/msm/camera_v2/isp/msm_isp_util.o
  CC      drivers/media/platform/msm/vidc/msm_v4l2_vidc.o
  CC      net/netfilter/xt_mark.o
In file included from drivers/media/platform/msm/camera_v2/isp/msm_isp_util.c:24:0:
include/trace/events/msm_cam.h:19:21: fatal error: msm_isp.h: No such file or directory
 #include "msm_isp.h"
                     ^
compilation terminated.
scripts/Makefile.build:258: recipe for target 'drivers/media/platform/msm/camera_v2/isp/msm_isp_util.o' failed
make[6]: *** [drivers/media/platform/msm/camera_v2/isp/msm_isp_util.o] Error 1
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm/camera_v2/isp' failed
make[5]: *** [drivers/media/platform/msm/camera_v2/isp] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm/camera_v2' failed
make[4]: *** [drivers/media/platform/msm/camera_v2] Error 2
make[4]: *** Waiting for unfinished jobs....
net/netfilter/xt_mark.c:16:37: fatal error: linux/netfilter/xt_mark.h: No such file or directory
 #include <linux/netfilter/xt_mark.h>
                                     ^
compilation terminated.
scripts/Makefile.build:258: recipe for target 'net/netfilter/xt_mark.o' failed
make[2]: *** [net/netfilter/xt_mark.o] Error 1
scripts/Makefile.build:403: recipe for target 'net/netfilter' failed
make[1]: *** [net/netfilter] Error 2
Makefile:967: recipe for target 'net' failed
make: *** [net] Error 2
make: *** Waiting for unfinished jobs....
  CC      drivers/media/platform/msm/vidc/msm_vidc_common.o
  CC      drivers/media/platform/msm/vidc/msm_vidc.o
  CC      drivers/media/platform/msm/vidc/msm_vdec.o
  CC      drivers/net/ethernet/msm/rndis_ipa.o
In file included from drivers/net/ethernet/msm/rndis_ipa_trace.h:81:0,
                 from drivers/net/ethernet/msm/rndis_ipa.c:32:
include/trace/define_trace.h:83:43: fatal error: ./rndis_ipa_trace.h: No such file or directory
 #include TRACE_INCLUDE(TRACE_INCLUDE_FILE)
                                           ^
compilation terminated.
scripts/Makefile.build:258: recipe for target 'drivers/net/ethernet/msm/rndis_ipa.o' failed
make[4]: *** [drivers/net/ethernet/msm/rndis_ipa.o] Error 1
scripts/Makefile.build:403: recipe for target 'drivers/net/ethernet/msm' failed
make[3]: *** [drivers/net/ethernet/msm] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/net/ethernet' failed
make[2]: *** [drivers/net/ethernet] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/net' failed
make[1]: *** [drivers/net] Error 2
make[1]: *** Waiting for unfinished jobs....
  LD      drivers/media/media.o
  CC      drivers/media/platform/msm/vidc/msm_venc.o
  CC      drivers/media/platform/msm/vidc/msm_smem.o
  CC      drivers/media/platform/msm/vidc/msm_vidc_debug.o
  CC      drivers/media/platform/msm/vidc/msm_vidc_res_parse.o
  CC      drivers/media/platform/msm/vidc/venus_hfi.o
  CC      drivers/media/platform/msm/vidc/hfi_response_handler.o
  CC      drivers/media/platform/msm/vidc/hfi_packetization.o
  CC      drivers/media/platform/msm/vidc/vidc_hfi.o
  CC      drivers/media/platform/msm/vidc/venus_boot.o
  CC      drivers/media/platform/msm/vidc/msm_vidc_dcvs.o
  CC      drivers/media/platform/msm/vidc/governors/msm_vidc_dyn_gov.o
  CC      drivers/media/platform/msm/vidc/governors/msm_vidc_table_gov.o
  CC      drivers/media/platform/msm/vidc/vmem/vmem.o
  CC      drivers/media/platform/msm/vidc/vmem/vmem_debugfs.o
  LD      drivers/media/platform/msm/vidc/governors/built-in.o
  LD      drivers/media/platform/msm/vidc/vmem/built-in.o
  LD      drivers/media/platform/msm/vidc/built-in.o
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm' failed
make[3]: *** [drivers/media/platform/msm] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media/platform' failed
make[2]: *** [drivers/media/platform] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media' failed
make[1]: *** [drivers/media] Error 2
Makefile:967: recipe for target 'drivers' failed
make: *** [drivers] Error 2
make: *** wait: No child processes.  Stop.

`

Gah0, 2018-12-02:
you should disable some config
or run  `make -d` find the issue

Z5X67280, 2018-12-02:
Do you have 'make O=out' ?

artyom5613, 2018-12-02:
Thanks for reply!
> you should disable some config
> or run `make -d` find the issue

Do you successively compile source code without any modifications? If yes what config you disable?
I write what i do,maybe something wrong: 1.Unpack source code. 2. Go to the arch/arm64/configs. copy config daisy.. in main source code folder. Then copy aarch-linux-android-4.9 into this folder. 3. cd in terminal to toolchain , write export CROSS_COMPILE=$(pwd)/bin/aarch64-linux-android-
 and export ARCH=arm64 && export SUBARCH=arm64 . Go to the main folder by cd in terminal and start conpiling. make daisy...      and make -j4 or make.

sushant9096, 2018-12-02:
Use make mrproper command then compile again

artyom5613, 2018-12-02:
> Use make mrproper command then compile again
I trying... but no effect.  they compile long time and then i got errors

sushant9096, 2018-12-02:
Show me error I can help you

artyom5613, 2018-12-02:
> Show me error I can help you

I can show you all log

sushant9096, 2018-12-02:
Okay

sushant9096, 2018-12-02:
Give me log

artyom5613, 2018-12-02:
> Give me log

[https://raw.githubusercontent.com/artyom5613/logfile/master/README.md](url)

sushant9096, 2018-12-02:
talk with me on telegram or give your log on hastening.com or del.dog and give here url

artyom5613, 2018-12-02:
https://del.dog/raw/povucibiru

sushant9096, 2018-12-02:
Use nano drivers/media/platform/msm/camera_v2/isp/msm_isp_util.c and modify # include "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h

artyom5613, 2018-12-02:
> Use nano drivers/media/platform/msm/camera_v2/isp/msm_isp_util.c and modify # include "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h

You mean go into file drivers/media/platform/msm/camera_v2/isp/msm_isp_util.c  and add string   # include "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h? I add this and now waiting for compiling...

sushant9096, 2018-12-02:
Yes modify this line # include "msm_isp.h" with this "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h"

artyom5613, 2018-12-02:
> Yes modify this line # include "msm_isp.h" with this "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h"

[https://github.com/MiCode/Xiaomi_Kernel_OpenSource/blob/daisy-o-oss/drivers/media/platform/msm/camera_v2/isp/msm_isp_util.c](url)
No line #include "msm_isp.h"

sushant9096, 2018-12-02:
In include/trace/events/msm_cam.h # include "msm_isp.h" modify that line with #  include "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h"

sushant9096, 2018-12-02:
Is the issue is solved..?

artyom5613, 2018-12-02:
> Is the issue is solved..?
Done what you write. I wait for full compilation.

sushant9096, 2018-12-02:
Remove that # include "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h" line from msm_isp_util.c otherwise you got no such file or directory error

sushant9096, 2018-12-02:
Sorry for my typing mistake

artyom5613, 2018-12-02:
> Remove that # include "../drivers/media/platform/msm/camera_v2/isp/msm_isp.h" line from msm_isp_util.c otherwise you got no such file or directory error

Yes, i remove. this issue solved. But others
.. In file included from drivers/media/platform/msm/camera_v2/sensor/io/msm_camera_tz_i2c.c:20:0:
drivers/media/platform/msm/camera_v2/sensor/msm_sensor.h:34:28: fatal error: msm_camera_i2c.h: No such file or directory
 #include "msm_camera_i2c.h"
                            ^
compilation terminated.
scripts/Makefile.build:258: recipe for target 'drivers/media/platform/msm/camera_v2/sensor/io/msm_camera_tz_i2c.o' failed
make[7]: *** [drivers/media/platform/msm/camera_v2/sensor/io/msm_camera_tz_i2c.o] Error 1
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm/camera_v2/sensor/io' failed
make[6]: *** [drivers/media/platform/msm/camera_v2/sensor/io] Error 2
make[6]: *** Waiting for unfinished jobs....
  CC      net/l2tp/l2tp_core.o
  CC      net/ipv6/netfilter/nf_conntrack_reasm.o
  LD      drivers/media/platform/msm/camera_v2/sensor/flash/built-in.o
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm/camera_v2/sensor' failed
make[5]: *** [drivers/media/platform/msm/camera_v2/sensor] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm/camera_v2' failed
make[4]: *** [drivers/media/platform/msm/camera_v2] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media/platform/msm' failed
make[3]: *** [drivers/media/platform/msm] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media/platform' failed
make[2]: *** [drivers/media/platform] Error 2
scripts/Makefile.build:403: recipe for target 'drivers/media' failed
make[1]: *** [drivers/media] Error 2
Makefile:967: recipe for target 'drivers' failed
make: *** [drivers] Error 2
make: *** Waiting for unfinished jobs....

artyom5613, 2018-12-03:
I trying reinstall full ubuntu(not vm not) But still this effect. Developers! Please fix issues, or provide instruction how to compile this!

closed 2018-12-03 by artyom5613

fsct2k, 2019-07-14:
> drivers/media/platform/msm/camera_v2/sensor/msm_sensor.h modify #include "msm_camera_i2c.h this with #include "io/msm_camera_i2c.h"

do you solved this, bro?

maximus-sallam, 2024-01-27:
same problem. when I copy include/uapi/linux/netfilter/xt_MARK.h to include/linux/netfilter/xt_mark.h I get this error instead:

  CC      net/netfilter/xt_mark.o
../net/netfilter/xt_mark.c:33:32: error: incomplete definition of type 'struct xt_mark_tginfo2'
        skb->mark = (skb->mark & ~info->mask) ^ info->mark;
                                  ~~~~^
../net/netfilter/xt_mark.c:31:15: note: forward declaration of 'struct xt_mark_tginfo2'
        const struct xt_mark_tginfo2 *info = par->targinfo;
                     ^
../net/netfilter/xt_mark.c:33:46: error: incomplete definition of type 'struct xt_mark_tginfo2'
        skb->mark = (skb->mark & ~info->mask) ^ info->mark;
                                                ~~~~^
../net/netfilter/xt_mark.c:31:15: note: forward declaration of 'struct xt_mark_tginfo2'
        const struct xt_mark_tginfo2 *info = par->targinfo;
                     ^
../net/netfilter/xt_mark.c:42:27: error: incomplete definition of type 'struct xt_mark_mtinfo1'
        return ((skb->mark & info->mask) == info->mark) ^ info->invert;
                             ~~~~^
../net/netfilter/xt_mark.c:40:15: note: forward declaration of 'struct xt_mark_mtinfo1'
        const struct xt_mark_mtinfo1 *info = par->matchinfo;
                     ^
../net/netfilter/xt_mark.c:42:42: error: incomplete definition of type 'struct xt_mark_mtinfo1'
        return ((skb->mark & info->mask) == info->mark) ^ info->invert;
                                            ~~~~^
../net/netfilter/xt_mark.c:40:15: note: forward declaration of 'struct xt_mark_mtinfo1'
        const struct xt_mark_mtinfo1 *info = par->matchinfo;
                     ^
../net/netfilter/xt_mark.c:42:56: error: incomplete definition of type 'struct xt_mark_mtinfo1'
        return ((skb->mark & info->mask) == info->mark) ^ info->invert;
                                                          ~~~~^
../net/netfilter/xt_mark.c:40:15: note: forward declaration of 'struct xt_mark_mtinfo1'
        const struct xt_mark_mtinfo1 *info = par->matchinfo;
                     ^
../net/netfilter/xt_mark.c:50:20: error: invalid application of 'sizeof' to an incomplete type 'struct xt_mark_tginfo2'
        .targetsize     = sizeof(struct xt_mark_tginfo2),
                          ^     ~~~~~~~~~~~~~~~~~~~~~~~~
../net/netfilter/xt_mark.c:50:34: note: forward declaration of 'struct xt_mark_tginfo2'
        .targetsize     = sizeof(struct xt_mark_tginfo2),
                                        ^
../net/netfilter/xt_mark.c:59:20: error: invalid application of 'sizeof' to an incomplete type 'struct xt_mark_mtinfo1'
        .matchsize      = sizeof(struct xt_mark_mtinfo1),
                          ^     ~~~~~~~~~~~~~~~~~~~~~~~~
../net/netfilter/xt_mark.c:59:34: note: forward declaration of 'struct xt_mark_mtinfo1'
        .matchsize      = sizeof(struct xt_mark_mtinfo1),
                                        ^
7 errors generated.
../scripts/Makefile.build:356: recipe for target 'net/netfilter/xt_mark.o' failed
make[3]: *** [net/netfilter/xt_mark.o] Error 1

maximus-sallam, 2024-01-27:
It looks like the xt_mark.h file is missing definitions:

root@ubuntu18:/home/kali/kernel/include/linux/netfilter# cat xt_mark.h
/* SPDX-License-Identifier: GPL-2.0 WITH Linux-syscall-note */
#ifndef _XT_MARK_H_target
#define _XT_MARK_H_target

#include <linux/netfilter/xt_mark.h>

#endif /*_XT_MARK_H_target */

maximus-sallam, 2024-01-27:
root@ubuntu18:/home/user/kernel# find . -type f -iname xt_mark*
./net/netfilter/xt_mark.c = original
./include/linux/netfilter/xt_MARK.h = I added this 
./include/linux/netfilter/xt_mark.h = I added this
./include/uapi/linux/netfilter/xt_MARK.h = original
./include/uapi/linux/netfilter/xt_mark.h = I added this

maximus-sallam, 2024-01-27:
I fixed it:

```
rm ./include/linux/netfilter/xt_MARK.h 
rm ./include/linux/netfilter/xt_mark.h
rm ./include/uapi/linux/netfilter/xt_mark.h
vim ./include/uapi/linux/netfilter/xt_mark.h

root@ubuntu18:/home/user/kernel/include/uapi/linux/netfilter# cat xt_mark.h
/* SPDX-License-Identifier: GPL-2.0 WITH Linux-syscall-note */
#ifndef _XT_MARK_H
#define _XT_MARK_H

#include <linux/types.h>

struct xt_mark_tginfo2 {
       __u32 mark, mask;
};

struct xt_mark_mtinfo1 {
       __u32 mark, mask;
       __u8 invert;
};

#endif /*_XT_MARK_H*/

root@ubuntu18:/home/user/kernel/include/uapi/linux/netfilter# diff -y xt_mark.h xt_MARK.h --suppress-common-lines
#ifndef _XT_MARK_H					      |	#ifndef _XT_MARK_H_target
#define _XT_MARK_H					      |	#define _XT_MARK_H_target
#include <linux/types.h>				      |	#include <linux/netfilter/xt_mark.h>
struct xt_mark_tginfo2 {				      |	#endif /*_XT_MARK_H_target */
       __u32 mark, mask;				      <
};							      <
							      <
struct xt_mark_mtinfo1 {				      <
       __u32 mark, mask;				      <
       __u8 invert;					      <
};							      <
							      <
#endif /*_XT_MARK_H*/					      <
```

maximus-sallam, 2024-01-27:
https://forums.developer.nvidia.com/t/error-while-compiling-kernel-with-docker-ubuntu-18-04-image/168604/3