MiCode/Xiaomi_Kernel_OpenSource #986: What i doing wrong when trying compiling?

artyom5613, 2018-12-08:
I trying compiling daisy-o-oss , scorpio-m-oss , but i face with errors(compilation successively started and go. But after 5-8 min i take error.  ).
Typical error:
```
 In file included from drivers/media/platform/msm/camera_v2/isp/msm_isp_util.c:24:0:
include/trace/events/msm_cam.h:19:21: fatal error: msm_isp.h: No such file or directory
 #include "msm_isp.h"
                     ^
```

What i do wrong? Download kernel source code->Download toolchain -> cd to the toolchain , type  export ARCH=arm64 && export SUBARCH=arm64 ->  export CROSS_COMPILE=$(pwd)/bin/aarch64-linux-android- -> cd to the source code -> make clean -> make mrproper -> make -j4.

Can you help me with instruction if I do something wrong?

OS: Ubuntu 18.04.1LTS. Maybe i need to try compile on another OS?

rama982, 2018-12-08:
Specify output dir with make O=out.

Revanth, 2018-12-09:
`export ARCH=arm64
export CROSS_COMPILE=~/Toolchain/bin/aarch64-linux-android-
mkdir output
make -C $(pwd) O=output rosy_defconfig
make -j32 -C $(pwd) O=output`

Revanth, 2018-12-09:
@artyom5613 U need to make a folder called as out. Posting the build Script in HasteBin have a look

[https://hastebin.com/idunehujat.js](Click Here]

ivan19871002, 2018-12-10:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/987

0xXA, 2018-12-10:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/wiki/How-to-compile-kernel-standalone

Maybe you'll get some help from wiki page

Moreover, you can paste link to a log file here if it really have anything to do with kernel

artyom5613, 2018-12-10:
thank you! It's help

closed 2018-12-10 by artyom5613