MiCode/Xiaomi_Kernel_OpenSource #992: MITV 3 60'' kernel source

saalmaan, 2018-12-22:
Hello, 

Can you please release MITV 3 kernel source code. Its been years since new update has arrived, very frustrating that we are stuck with Android 5.1. With the release of this Kernel source code, developer can at least bring latest Android TV to MITV 3 and give this TV new lease of life. 
If this is not the right forum for asking kernel source for MITV, then please direct me to correct forum.
Thanks

saalmaan, 2019-04-01:
@ivan19871002 Any update on this? You mentioned on Weibo that MITV source kernel is coming soon, do you know any approximate dates? https://m.weibo.cn/detail/4341835327348409

ivan19871002, 2019-04-02:
https://github.com/MiCode/MiBox_Kernel_OpenSource
this repo

if you have some question, pls create issue

closed 2019-04-02 by ivan19871002

saalmaan, 2019-04-02:
@ivan19871002 Thank you for replying. Repo you linked is MiBox, my request was for MiTV 3, Can you please help with source kernel for this? 

Or you are suggesting that MIBOX3(global) kernel can be used for MiTV 3 60''?